CadQuery/CQ-editor #270: Segmentation fault when rendering a cadquery script that fillets edges

safdark, 2021-06-09:
## To Reproduce

Here's the cadquery script I am using. Rendering this causes CQ-Editor to segfault.

```python
import cadquery as cq
start_radius = 30
slot_size = 7
slot_inset = 10

length = 100
end_radius = 20

outer = cq.Workplane("XY").circle(start_radius)
lofted = outer.workplane(offset=length).circle(end_radius).loft(combine=True)
channeled = lofted.faces(">Z").rect((2 * end_radius - 3 * slot_inset), (2 * end_radius - 3 * slot_inset), forConstruction=True)\
           .vertices().rect(slot_size, slot_size).cutThruAll()\
            .faces("|Z").edges().fillet(3) # <---- This line causes the segmentation fault
```

## Environment

OS: Macbook 10.15.7

Was CadQuery installed using Conda?: Yes

Output of `conda list` from your active Conda environment:
```
# packages in environment at /Users/blah/anaconda3/envs/cad:
#
# Name                    Version                   Build  Channel
alabaster                 0.7.12                     py_0    conda-forge
appdirs                   1.4.4              pyh9f0ad1d_0    conda-forge
applaunchservices         0.2.1                      py_0    conda-forge
appnope                   0.1.2            py38h50d1736_1    conda-forge
argh                      0.26.2          pyh9f0ad1d_1002    conda-forge
astroid                   2.5.6            py38h50d1736_0    conda-forge
async_generator           1.10                       py_0    conda-forge
atomicwrites              1.4.0              pyh9f0ad1d_0    conda-forge
attrs                     21.2.0             pyhd8ed1ab_0    conda-forge
autopep8                  1.5.6              pyhd8ed1ab_0    conda-forge
babel                     2.9.1              pyh44b312d_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
black                     21.5b1             pyhd8ed1ab_0    conda-forge
bleach                    3.3.0              pyh44b312d_0    conda-forge
brotlipy                  0.7.0           py38h5406a74_1001    conda-forge
ca-certificates           2021.5.30            h033912b_0    conda-forge
cadquery                  2.1                       py3.8    cadquery
certifi                   2021.5.30        py38h50d1736_0    conda-forge
cffi                      1.14.5           py38ha97d567_0    conda-forge
chardet                   4.0.0            py38h50d1736_1    conda-forge
click                     8.0.1            py38h50d1736_0    conda-forge
cloudpickle               1.6.0                      py_0    conda-forge
colorama                  0.4.4              pyh9f0ad1d_0    conda-forge
cq-editor                 0.2                       py3.8    cadquery
cryptography              3.4.7            py38h1fa4640_0    conda-forge
dataclasses               0.8                pyhc8e2a94_1    conda-forge
dbus                      1.13.16              h18a8e69_0  
decorator                 5.0.9              pyhd8ed1ab_0    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
diff-match-patch          20200713           pyh9f0ad1d_0    conda-forge
docutils                  0.17.1           py38h50d1736_0    conda-forge
entrypoints               0.3             pyhd8ed1ab_1003    conda-forge
expat                     2.2.9                h4a8c4bd_2    conda-forge
ezdxf                     0.14.2                     py_0    conda-forge
flake8                    3.8.4                      py_0    conda-forge
fontconfig                2.13.1            h10f422b_1005    conda-forge
freetype                  2.10.4               h4cff582_1    conda-forge
[... 118 lines cut ...]
testpath                  0.5.0              pyhd8ed1ab_0    conda-forge
textdistance              4.2.1              pyhd8ed1ab_0    conda-forge
three-merge               0.1.1              pyh9f0ad1d_0    conda-forge
tk                        8.6.10               hb0a8c7a_0  
toml                      0.10.2             pyhd8ed1ab_0    conda-forge
tornado                   6.1              py38h5406a74_1    conda-forge
traitlets                 5.0.5                      py_0    conda-forge
typed-ast                 1.4.3            py38h96a0964_0    conda-forge
typing_extensions         3.7.4.3                    py_0    conda-forge
typish                    1.9.2              pyhd8ed1ab_0    conda-forge
ujson                     3.0.0            py38hc84c608_0    conda-forge
urllib3                   1.26.5             pyhd8ed1ab_0    conda-forge
watchdog                  2.1.2            py38h3c40b66_0    conda-forge
wcwidth                   0.2.5              pyh9f0ad1d_2    conda-forge
webencodings              0.5.1                      py_1    conda-forge
wheel                     0.36.2             pyhd3eb1b0_0  
wrapt                     1.12.1           py38h5406a74_3    conda-forge
wurlitzer                 2.1.0            py38h50d1736_0    conda-forge
xz                        5.2.5                h1de35cc_0  
yaml                      0.2.5                haf1e3a3_0    conda-forge
yapf                      0.31.0             pyhd8ed1ab_0    conda-forge
zeromq                    4.3.2                h6de7cb9_2    conda-forge
zipp                      3.4.1              pyhd8ed1ab_0    conda-forge
zlib                      1.2.11               h1de35cc_3  
```

safdark, 2021-06-09:
Thanks for the prompt reply!

I'll try that out. Should this be logged as an issue wth the OCCT CAD kernel then? Segmentation faults like this, instead of user-friendly parse/compile errors, impact the usability of tools higher up the chain too, such as cadquery and cq-editor.

marcus7070, 2021-06-10:
Logging bugs upstream is more complicated. See https://dev.opencascade.org/doc/overview/html/occt_contribution__contribution_workflow.html#occt_contribution_workflow_issue

You have to reproduce the error in OCCT's Drawexe. I believe you also have to sign an agreement and set up an account.

It would also be worth searching through their bug tracker. I think it's likely there is already a bug for this issue: https://tracker.dev.opencascade.org/view_all_bug_page.php