CadQuery/CQ-editor #274: New Release

jmwright, 2021-07-02:
Can we do a 0.2.1 release of CQ-editor? I've got pre-built packages on my fork with the latest version that I and other users are having good luck with. I would be happy to draft a new release and attach the archives to it so that it's easier for users to find and download.

adam-urbanczyk, 2021-07-02:
The release of CQ-editor is related to the release of CQ

jmwright, 2021-07-02:
Then can we do like I have with cq-cli and do a 2.2-beta.1 release?

adam-urbanczyk, 2021-07-02:
I don't think that this is sustainable - tomorrow there will be someone with yet another issue that is not present in master.

Do you know where we could upload nightly builds from your pyinstaller pipeline - if possible that would be IMO much better?

jmwright, 2021-07-02:
I have the schedule set to build weekly right now, but I can do nightly if that will help. I'm not sure if GitHub changed something or not, but I thought that the builds on my fork were only visible if you were logged into GitHub. It appears that's not the case though as I can see these from a browser where I'm not logged into GitHub. These are still mentioned in the CQ-editor readme. https://github.com/jmwright/CQ-editor/actions/workflows/pyinstaller-builds-actions.yml

narvalotech, 2023-07-22:
> The release of CQ-editor is related to the release of CQ

Can what @jmwright said be implemented? ie have a latest/unstable/nightly release version automatically updated by the CI builds? Because every time this is mentioned, the instructions are to navigate the CI pipeline page to find and download the artifacts. IMO that's just adding unnecessary friction, most users will expect the docs page to be aligned with the latest release.

I'm using a version installed from the AUR and it grabs the latest GH release, but that version is out of date with the docs, just found out now when it refused to add a `.sketch()` to my workplane.

If it's a problem of time, I can help out.

adam-urbanczyk, 2023-07-22:
Are you looking for this: https://github.com/CadQuery/CQ-editor/releases/tag/nightly ?

narvalotech, 2023-07-22:
Ah yeah I feel dumb now 😅

I still think it's not clear enough though, maybe making that tag more visible somewhere in the docs or on the readme:

could be in the documentation directly for features that are not supported on the stable release (ie Sketch) or there could be in the readme of cq-editor a section on the features that are missing from the stable release.

It seems the problem is that the stable cq-editor releases are tied to cq itself, right? Is there a particular reason that a stable cq release hasnt been cut?
I'm running a nightly package (from the ci artifacts) on my other laptop (Ubuntu) and it doesnt seem particularly unstable.

In any case, Im really happy about both cq and its editor that you made! 
Just want to make the install smoother, thats all.

closed 2025-02-06 by jmwright