CadQuery/CQ-editor #275: cq-editor console fails with input lines with broken escape sequences

carribeiro, 2021-07-11:
When a string with a broken escape sequence is input into console, cq-editor freezes for a moment (about 15 seconds in my PC).  After some time it unfreezes and shows a log in the log viewer. However the console becomes unusable after that until I restart cq-editor. Lines are not interpreted anymore..

### Sequence on the console (steps to reproduce)

```
In [1]: s = "abc"

In [2]:print(s)
abc

In [3]: s = "C:\Users\Atual\Google Drive\"
## freezes forabout 15 seconds, no output ##

In [4]: print(s)
## freezes forabout 15 seconds, no output ##

In [5]: s = "abc"
## freezes again ##

In [5]:print(s)
## freezes again, no output ##
```

### Log Viewer output

```
[2021-07-11 17:59:27.404491] ERROR: tornado.general: Uncaught exception in ZMQStream callback
Traceback (most recent call last):
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\zmq\eventloop\zmqstream.py", line 434, in _run_callback
    callback(*args, **kwargs)
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\ipykernel\iostream.py", line 126, in _handle_event
    event_f()
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\ipykernel\iostream.py", line 223, in <lambda>
    self.schedule(lambda : self._really_send(*args, **kwargs))
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\ipykernel\iostream.py", line 231, in _really_send
    self.socket.send_multipart(msg, *args, **kwargs)
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\ipykernel\inprocess\socket.py", line 37, in send_multipart
    self.message_sent += 1
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\traitlets\traitlets.py", line 604, in __set__
    self.set(obj, value)
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\traitlets\traitlets.py", line 593, in set
    obj._notify_trait(self.name, old_value, new_value)
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\traitlets\traitlets.py", line 1217, in _notify_trait
    self.notify_change(Bunch(
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\traitlets\traitlets.py", line 1227, in notify_change
    return self._notify_observers(change)
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\traitlets\traitlets.py", line 1264, in _notify_observers
    c(event)
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\ipykernel\inprocess\ipkernel.py", line 134, in _io_dispatch
    ident, msg = self.session.recv(self.iopub_socket, copy=False)
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\jupyter_client\session.py", line 809, in recv
    msg_list = socket.recv_multipart(mode, copy=copy)
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\ipykernel\iostream.py", line 255, in __getattr__
    warnings.warn("Accessing zmq Socket attribute %s on BackgroundSocket" % attr,
DeprecationWarning: Accessing zmq Socket attribute recv_multipart on BackgroundSocket
[2021-07-11 17:59:27.477095] ERROR: tornado.general: Uncaught exception in zmqstream callback
Traceback (most recent call last):
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\zmq\eventloop\zmqstream.py", line 451, in _handle_events
    self._handle_recv()
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\zmq\eventloop\zmqstream.py", line 480, in _handle_recv
    self._run_callback(callback, msg)
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\zmq\eventloop\zmqstream.py", line 434, in _run_callback
    callback(*args, **kwargs)
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\ipykernel\iostream.py", line 126, in _handle_event
    event_f()
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\ipykernel\iostream.py", line 223, in <lambda>
    self.schedule(lambda : self._really_send(*args, **kwargs))
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\ipykernel\iostream.py", line 231, in _really_send
    self.socket.send_multipart(msg, *args, **kwargs)
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\ipykernel\inprocess\socket.py", line 37, in send_multipart
[... 59 lines cut ...]
DeprecationWarning: Accessing zmq Socket attribute recv_multipart on BackgroundSocket
[2021-07-11 17:59:47.534443] INFO: ipykernel.inprocess.ipkernel: Exception in execute request:
[1;36m  File [1;32m"C:\Users\Atual\AppData\Local\Temp/ipykernel_9144/1192934994.py"[1;36m, line [1;32m1[0m
[1;33m    s = "C:\Users\Atual\Google Drive\"[0m
[1;37m                                      ^[0m
[1;31mSyntaxError[0m[1;31m:[0m EOL while scanning string literal
```

### Conda terminal window output

On the conda terminal window, the following sequence of errors is registered:

```
(cq) C:\Users\Atual>cq-editor
Namespace(filename=None)
IOStream.flush timed out
IOStream.flush timed out
...
```

### Version info

Version: 0.3.0dev
Windows 10
Installed using Conda and Python 3.9.1

adam-urbanczyk, 2021-07-11:
I get a SyntaxError on Linux, will check on Windows

```s = "C:\Users\Atual\Google Drive\"
  File "<ipython-input-8-7f5b0da71207>", line 1
    s = "C:\Users\Atual\Google Drive\"
                                      ^
SyntaxError: EOL while scanning string literal
```

carribeiro, 2021-07-11:
In fact, it HAS to issue a SyntaxError because we're feeding a broken statement to the interpreter with unbalanced quotes.

The problem is in the handling of the error. If you try the same on IPython, it gives the same message (of course as it is really a syntax error), but it keeps working. In this case something breaks inside cq-editor (at least on Windows) and it stops working to the point where you need to close & reopen cq-editor.

adam-urbanczyk, 2021-07-12:
I cannot reproduce on windows - I just get a SyntaxError without the extra stuff. Please share your env specification (`conda list`).

carribeiro, 2021-07-12:
```
(cq) C:\Users\Atual>conda list
# packages in environment at C:\Users\Atual\miniconda3\envs\cq:
#
# Name                    Version                   Build  Channel
alabaster                 0.7.12                     py_0    conda-forge
appdirs                   1.4.4              pyh9f0ad1d_0    conda-forge
argh                      0.26.2          pyh9f0ad1d_1002    conda-forge
astroid                   2.6.2            py39hcbf5309_0    conda-forge
async_generator           1.10                       py_0    conda-forge
atomicwrites              1.4.0              pyh9f0ad1d_0    conda-forge
attrs                     21.2.0             pyhd8ed1ab_0    conda-forge
autopep8                  1.5.6              pyhd8ed1ab_0    conda-forge
babel                     2.9.1              pyh44b312d_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
bcrypt                    3.2.0            py39hb82d6ee_1    conda-forge
black                     21.5b2             pyhd8ed1ab_0    conda-forge
bleach                    3.3.0              pyh44b312d_0    conda-forge
brotlipy                  0.7.0           py39hb82d6ee_1001    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2021.5.30            h5b45459_0    conda-forge
cadquery                  master                    py3.9    cadquery
certifi                   2021.5.30        py39hcbf5309_0    conda-forge
cffi                      1.14.6           py39h0878f49_0    conda-forge
chardet                   4.0.0            py39hcbf5309_1    conda-forge
click                     8.0.1            py39hcbf5309_0    conda-forge
cloudpickle               1.6.0                      py_0    conda-forge
colorama                  0.4.4              pyh9f0ad1d_0    conda-forge
cq-editor                 master                    py3.9    cadquery
cryptography              3.4.7            py39hd8d06c1_0    conda-forge
curl                      7.77.0               h789b8ee_0    conda-forge
dataclasses               0.8                pyhc8e2a94_1    conda-forge
debugpy                   1.3.0            py39h415ef7b_0    conda-forge
decorator                 5.0.9              pyhd8ed1ab_0    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
diff-match-patch          20200713           pyh9f0ad1d_0    conda-forge
docutils                  0.17.1           py39hcbf5309_0    conda-forge
double-conversion         3.1.5                h0e60522_2    conda-forge
eigen                     3.3.9                h2d74725_1    conda-forge
entrypoints               0.3             pyhd8ed1ab_1003    conda-forge
expat                     2.4.1                h39d44d4_0    conda-forge
ezdxf                     0.16.4           py39h2e07f2f_0    conda-forge
ffmpeg                    4.3.1                ha925a31_0    conda-forge
flake8                    3.8.4                      py_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.13.1            h1989441_1005    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freeimage                 3.18.0               hfcb8cd7_7    conda-forge
freetype                  2.10.4               h546665d_1    conda-forge
future                    0.18.2           py39hcbf5309_3    conda-forge
gl2ps                     1.4.2                h0597ee9_0    conda-forge
glew                      2.1.0                h39d44d4_2    conda-forge
hdf4                      4.2.15               h0e5069d_3    conda-forge
hdf5                      1.10.6          nompi_h5268f04_1114    conda-forge
icu                       68.1                 h0e60522_0    conda-forge
idna                      2.10               pyh9f0ad1d_0    conda-forge
ilmbase                   2.5.5                h12d4b20_0    conda-forge
imagesize                 1.2.0                      py_0    conda-forge
importlib-metadata        4.6.1            py39hcbf5309_0    conda-forge
importlib_metadata        4.6.1                hd8ed1ab_0    conda-forge
intel-openmp              2021.3.0          h57928b3_3372    conda-forge
intervaltree              3.0.2                      py_0    conda-forge
ipykernel                 6.0.1            py39h832f523_0    conda-forge
ipython                   7.25.0           py39h832f523_1    conda-forge
[... 138 lines cut ...]
typing_extensions         3.10.0.0           pyha770c72_0    conda-forge
typish                    1.9.2              pyhd8ed1ab_0    conda-forge
tzdata                    2021a                h52ac0ba_0
ujson                     4.0.2            py39h415ef7b_0    conda-forge
urllib3                   1.26.6             pyhd8ed1ab_0    conda-forge
utfcpp                    3.2.1                h57928b3_0    conda-forge
vc                        14.2                 h21ff451_1
vs2015_runtime            14.27.29016          h5e58377_2
[... cut ...]
xz                        5.2.5                h62dcd97_1    conda-forge
yaml                      0.2.5                he774522_0    conda-forge
yapf                      0.31.0             pyhd8ed1ab_0    conda-forge
zeromq                    4.3.4                h0e60522_0    conda-forge
zipp                      3.5.0              pyhd8ed1ab_0    conda-forge
zlib                      1.2.11            h62dcd97_1010    conda-forge
zstd                      1.5.0                h6255e5f_0    conda-forge
```

carribeiro, 2021-07-12:
Just an additional remark: I installed cq-editor twice on this PC and got the same error both times. I deleted the entire env and reinstalled later, to the same result.

carribeiro, 2021-07-12:
After updating, now it works the same as for you. I was getting this problem for something like two weeks (since I've triggered it by mistake the first time). It was probably fixed somewhere upstream.

It's probably solved by now.

carribeiro, 2021-07-12:
Too soon: `1 +& 2` still triggers the error ( a bit different this time).

```
[2021-07-12 22:45:42.864755] ERROR: tornado.general: Uncaught exception in ZMQStream callback
Traceback (most recent call last):
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\zmq\eventloop\zmqstream.py", line 434, in _run_callback
    callback(*args, **kwargs)
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\ipykernel\iostream.py", line 126, in _handle_event
    event_f()
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\ipykernel\iostream.py", line 223, in <lambda>
    self.schedule(lambda : self._really_send(*args, **kwargs))
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\ipykernel\iostream.py", line 231, in _really_send
    self.socket.send_multipart(msg, *args, **kwargs)
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\ipykernel\inprocess\socket.py", line 37, in send_multipart
    self.message_sent += 1
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\traitlets\traitlets.py", line 604, in __set__
    self.set(obj, value)
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\traitlets\traitlets.py", line 593, in set
    obj._notify_trait(self.name, old_value, new_value)
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\traitlets\traitlets.py", line 1217, in _notify_trait
    self.notify_change(Bunch(
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\traitlets\traitlets.py", line 1227, in notify_change
    return self._notify_observers(change)
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\traitlets\traitlets.py", line 1264, in _notify_observers
    c(event)
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\ipykernel\inprocess\ipkernel.py", line 134, in _io_dispatch
    ident, msg = self.session.recv(self.iopub_socket, copy=False)
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\jupyter_client\session.py", line 809, in recv
    msg_list = socket.recv_multipart(mode, copy=copy)
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\ipykernel\iostream.py", line 255, in __getattr__
    warnings.warn("Accessing zmq Socket attribute %s on BackgroundSocket" % attr,
DeprecationWarning: Accessing zmq Socket attribute recv_multipart on BackgroundSocket
[2021-07-12 22:45:43.704287] ERROR: tornado.general: Uncaught exception in zmqstream callback
Traceback (most recent call last):
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\zmq\eventloop\zmqstream.py", line 451, in _handle_events
    self._handle_recv()
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\zmq\eventloop\zmqstream.py", line 480, in _handle_recv
    self._run_callback(callback, msg)
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\zmq\eventloop\zmqstream.py", line 434, in _run_callback
    callback(*args, **kwargs)
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\ipykernel\iostream.py", line 126, in _handle_event
    event_f()
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\ipykernel\iostream.py", line 223, in <lambda>
    self.schedule(lambda : self._really_send(*args, **kwargs))
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\ipykernel\iostream.py", line 231, in _really_send
    self.socket.send_multipart(msg, *args, **kwargs)
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\ipykernel\inprocess\socket.py", line 37, in send_multipart
    self.message_sent += 1
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\traitlets\traitlets.py", line 604, in __set__
    self.set(obj, value)
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\traitlets\traitlets.py", line 593, in set
    obj._notify_trait(self.name, old_value, new_value)
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\traitlets\traitlets.py", line 1217, in _notify_trait
    self.notify_change(Bunch(
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\traitlets\traitlets.py", line 1227, in notify_change
    return self._notify_observers(change)
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\traitlets\traitlets.py", line 1264, in _notify_observers
    c(event)
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\ipykernel\inprocess\ipkernel.py", line 134, in _io_dispatch
    ident, msg = self.session.recv(self.iopub_socket, copy=False)
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\jupyter_client\session.py", line 809, in recv
    msg_list = socket.recv_multipart(mode, copy=copy)
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\ipykernel\iostream.py", line 255, in __getattr__
    warnings.warn("Accessing zmq Socket attribute %s on BackgroundSocket" % attr,
DeprecationWarning: Accessing zmq Socket attribute recv_multipart on BackgroundSocket
[2021-07-12 22:45:43.903955] ERROR: asyncio: Exception in callback AddThreadSelectorEventLoop._handle_select([1184], [])
handle: <Handle AddThreadSelectorEventLoop._handle_select([1184], [])>
Traceback (most recent call last):
  File "C:\Users\Atual\miniconda3\envs\cq\lib\asyncio\events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
[... 18 lines cut ...]
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\ipykernel\inprocess\socket.py", line 37, in send_multipart
    self.message_sent += 1
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\traitlets\traitlets.py", line 604, in __set__
    self.set(obj, value)
[... cut ...]
  File "C:\Users\Atual\miniconda3\envs\cq\lib\site-packages\ipykernel\iostream.py", line 255, in __getattr__
    warnings.warn("Accessing zmq Socket attribute %s on BackgroundSocket" % attr,
DeprecationWarning: Accessing zmq Socket attribute recv_multipart on BackgroundSocket
[2021-07-12 22:46:04.374316] INFO: ipykernel.inprocess.ipkernel: Exception in execute request:
[1;36m  File [1;32m"C:\Users\Atual\AppData\Local\Temp/ipykernel_5476/2953139260.py"[1;36m, line [1;32m1[0m
[1;33m    1 +& 2[0m
[1;37m       ^[0m
[1;31mSyntaxError[0m[1;31m:[0m invalid syntax
```

carribeiro, 2021-07-12:
> I cannot reproduce on windows - I just get a SyntaxError without the extra stuff. Please share your env specification (`conda list`).

Try with this: `1 +& 2. ` 

It still gives me the same problem.

adam-urbanczyk, 2021-07-14:
I cannot reproduce this on my win PC.