CadQuery/CQ-editor #277: Resizer between the panels at 4k resolution very hard to grab

carribeiro, 2021-07-14:
Ok, that seems silly but it's really frustrating.

The resizer between the panels on cq-editor at very high resolutions (I have a 4k monitor) becomes VERY hard to use. It took me a few attempts to increase the size of the editor panel. The region where you can drag the divider to resize the panels is very small. I'm using a MacBook (OSX) and CadQuery up-to-date from master. I believe (but can't really test) that the issue may be limited to OSX.

adam-urbanczyk, 2021-07-14:
Any suggestions how to solve this (e.g. a qss style sheet)? I don't have such a display to investigate.

lorenzncode, 2021-08-19:
Found the style setting.  Here with a large value for testing.

main_window.py in MainWindow.__init__()
```
        stylesheet = """
        QMainWindow::separator {
            width:  40px;
            height: 40px;
            }
        """
        self.setStyleSheet(stylesheet)
```
[image]

lorenzncode, 2021-08-22:
VS Code provides a similar setting named "Sash: Size"
Default = 4, Maximum = 20

> Controls the feedback area size in pixels of the dragging area in between views/editors. Set it to a larger value if you feel it's hard to resize views using the mouse.

[image]

But they have special handling so the "feedback" area can be larger than the displayed size.

bendavis78, 2023-09-24:
@lorenzncode thanks, that fixed it for me (I found 5 to 10px to be a comfortable size) 👍