CadQuery/CQ-editor #278: Ability to reload with changes from external editor

andrewebdev, 2021-07-19:
Hi, I really like the visual feedback and quick debug info that CQ-editor provides.
That said, I reaaaaally prefer editing python code in my editor of choice (doom-emacs).

Currently I edit my code separately in emacs, and then switch over to CQ-editor to preview my changes. But I have to re-open the file every time so that the changes are synced, before I can preview it.

*What would be nice* is if CQ-editor can detect that the file changed, reloads the contents, and if (auto-reload) is selected in CQ-editor, then the preview is also updated.

jmwright, 2021-07-19:
There is an "automatic reload" setting for use with external editors in the Preferences dialog.

andrewebdev, 2021-07-19:
ah thx @jmwright my bad. I actually thought that was just to auto-reload the code for rendering the object, not auto reloading "from file"... my bad, should have just tested it :roll_eyes:

closed 2021-07-19 by andrewebdev