CadQuery/CQ-editor #280: Automatic reload and preview randomly stops working

r4dr3fr4d, 2021-08-05:
I have auto-reload enabled, but often when I write to my file in vim, CQ-editor doesn't pick up the changes.  Sometimes I can uncheck and re-check the option, write to the file again, and it will correct itself, other times I have to close and restart the program entirely.  Has anyone else observed this issue?  I'm on Arch Linux and am using the latest release build.

adam-urbanczyk, 2021-08-05:
Take a look here: https://github.com/CadQuery/CQ-editor/pull/48

Does changing the delay solve your issue?

r4dr3fr4d, 2021-08-06:
Thanks for the reply, but not sure how to test that with the release build.

marcus7070, 2021-08-09:
> Has anyone else observed this issue?

I use CQ-editor and Vim and have not noticed this on my laptop or desktop.

patniemeyer, 2025-03-08:
I just cloned the latest tonight (python 3.12 using python venv on MacOS) and I find that it only reloads *once*... I can launch the run.py with the filename as an arg and then edit in vim... On the first change it will update and then it just stops and nothing I find that I can do in the gui will get it to update.

jmwright, 2025-03-08:
@patniemeyer Some things to try. 

1. Can you try adjusting the `Autoreload delay` in the Editor preferences [link](https://github.com/CadQuery/CQ-editor/blob/master/README.md#using-an-external-code-editor)?
2. Does automatic reload work with the built-in Spyder editor? 
3. Does reload work with a different external editor (VS Code, a text editor, etc)?
4. Does it work if you open the file via `File->Open` instead?

patniemeyer, 2025-03-08:
So I tried setting the autoreload delay down to "2" (not sure if that is millis or seconds?) but it didn't seem to change the behavior either for the internal or external editor. 

I have autoreload turned on and I see it on as reflected in the button state and the checkbox in the preferences editor in the app.

It's a little confusing because I'm not sure if the internal editor is supposed to dynamically update or only when I hit "save"?    Upon launch the view is blank until I hit save.  Thereafter hitting save does update, but I've never seen it auto-reload other than the one-time thing when using an *external* editor.

As before, when modifying the external file in vim or another text editor I get exactly one update.  The update is reflected in the internal file editor view.

Some more background - I installed this on MacOS by cloning master from this repo, creating a venv with python 3.12, and using`pip install .`  I also tried installing the cadquery package via pip from git as shown in the README but that version seems to break the environment?

When I get some time soon I'll try to dig into the code and see if I can figure out what's going on...  Thanks.

jmwright, 2025-03-08:
> I also tried installing the cadquery package via pip from git as shown in the README but that version seems to break the environment?

How does it break the environment?

You can also just run `pip install CQ-editor` to get the latest release.

I have to borrow a Mac for testing, but I'll try to borrow one and see if I can reproduce this issue.

patniemeyer, 2025-03-08:
Installing with `pip install CQ-editor` solved the problem!  (Thanks).  It now auto-reloads upon changes to the file.  I'm not sure why I didn't start with that route...

jmwright, 2025-03-08:
Glad it works for you. I think the next step is to determine that it's not a regression introduced after the 0.4 release.

jmwright, 2025-03-09:
I confirmed that this is a regression since the 0.4 release. #502 is my attempt to fix that while preserving the added functionality that broke auto-reload for external editors in the first place.

@patniemeyer If you get bored, you could try that branch out and make sure it works properly for you too.

patniemeyer, 2025-03-09:
@jmwright 👍 The branch looks good. A clean install with pip works (reloads) for me.

By the way, just wanted to say that although I'm just getting started with it I'm super excited about cadquery.  I was a huge fan of OpenSCAD and used it for years but always found it frustrating and slow.  CadQuery is already bringing back some of that joy that I had at the beginning :)

jmwright, 2025-03-09:
@patniemeyer Glad to hear it! The fluent API (default way of programming with CadQuery) is great, but the newer [free function API](https://cadquery.readthedocs.io/en/latest/free-func.html) provides a very different way of using CadQuery that some users find more intuitive (there's no hidden state), and that can become more helpful with complex models. I'm not trying to push you that way, but be aware that you have that option to explore if you feel the need.

closed 2025-03-09 by jmwright