CadQuery/CQ-editor #281: Debug workplanes

revolter, 2021-08-14:
I feel like it would be very helpful if you could visualize workplanes, for example, using `debug`, or even `show_object`. Bonus points for visualizing the axis.

adam-urbanczyk, 2021-08-14:
`Tools->Inspect  CQ object` allows to visualize this

revolter, 2021-08-17:
I see. And could you please explain what do the items from the "CQ object inspector" section mean, and what does their order mean too?

lorenzncode, 2021-08-19:
Go through the CadQuery Introspective Example if you haven't already.  If you `print(part)` at each step and compare with the CQ-Editor you can see that each object has been added to the object inspector.

revolter, 2021-08-19:
With this example:

```py
import cadquery as cq

width = 30
height = 100
thickness = 2

handle = (
    cq
    .Workplane('front')
    .circle(width * 2)
    .extrude(height)

    .workplane()
    .tag('main')

    .copyWorkplane(
        cq
        .Workplane('bottom')
    )
    .split(keepTop=True)

    .copyWorkplane(
        cq
        .Workplane('right')
    )
    .split(keepTop=True)

    .faces('>Y or >Z or <Z')
    .shell(thickness)
    .fillet(thickness / 2 - 0.001)
)

show_object(
    handle,
    options={
        'alpha': 0.3
    }
)
```

If I select the first item from the bottom, this error is printed in the "Log viewer" pane:

```
[2021-08-19 05:22:13.857691] ERROR: CQ-Editor: Uncaught exception occurred
Traceback (most recent call last):
  File "/usr/local/Caskroom/miniconda/base/lib/python3.8/site-packages/cq_editor/widgets/cq_object_inspector.py", line 92, in handleSelection
    dim = item.workplane.largestDimension()
  File "/usr/local/Caskroom/miniconda/base/lib/python3.8/site-packages/cadquery/cq.py", line 2739, in largestDimension
    compound = self.findSolid()
  File "/usr/local/Caskroom/miniconda/base/lib/python3.8/site-packages/cadquery/cq.py", line 778, in findSolid
    raise ValueError(
ValueError: Cannot find a solid on the stack or in the parent chain
```

And then:
- the second one is a cylinder (correct)
- the third one is a cylinder (probably correct, though not clear why: maybe the `.workplane()`, maybe the `.tag()`, so it would be helpful if this could be improved to make it clearer)
- the fourth one is the handle (wrong)
- the fifth one is the handle (wrong)
- the sixth one is half of cylinder (correct)
and so forth.

This is why I was confused about the order, because it jumps from the intermediate steps to the final step randomly, while going through the list in order.

lorenzncode, 2021-08-20:
> And could you please explain what do the items from the "CQ object inspector" pane mean, and what does their order mean too?

The CQ object inspector lists the Workplane objects on the stack.  Each Workplane object has a list of objects (can be Wire, Compound, Face, Vertex and so on).  

The bottom item has the most parents (where you started the part).  The top item has no more parent objects (where you finished).

The Vector that is displayed for each item is the Workplane origin.

When the Workplane object list is non-empty, there is a triangle that you can select to expand the tree and display the objects.  When the Workplane objects list is empty, the tree cannot be expanded.  The Workplane origin is displayed in the Type column and the Value column is empty in the case of empty objects list.

In your example, the fourth and fifth are correct.  I'd suggest toggling the visibility of the handle model under Objects CQ models.  Inspecting a Workplane, from the CQ object inspector, that has an empty list of objects, does not display anything.

> If I select the first item from the bottom, this error is printed in the "Log viewer" pane

Yes I can reproduce the error, also with:
```
import cadquery as cq
r = cq.Workplane().circle(1)
```

Perhaps CQ-Editor should be changed to fix/suppress this error that results from findSolid() failing when inspecting objects that do not have a solid.

revolter, 2021-08-21:
> The bottom item has the most parents (where you started the part). The top item has no more parent objects (where you finished).

Isn't it the other way around? Meaning, the start part doesn't have any parent.

> I'd suggest toggling the visibility of the handle model under Objects CQ models.

This was it! Wouldn't it be a good idea to have it hide automatically when you select an "older" item? And when you select the last item in the list, which is the final object, it could unhide back.

> Perhaps CQ-Editor should be changed to fix/suppress this error that results from findSolid() failing when inspecting objects that do not have a solid.

Do you want me to create a separate issue for this?

---

I understood it better after your explanations, thank you!

But I do have one more question: when an intermediate workplane is displayed, how do I know what are its x, y and z axis oriented?

revolter, 2021-08-21:
I found another error:

```
[2021-08-21 08:33:03.435788] ERROR: CQ-Editor: Uncaught exception occurred
Traceback (most recent call last):
  File "/usr/local/Caskroom/miniconda/base/lib/python3.8/site-packages/cq_editor/widgets/cq_object_inspector.py", line 102, in handleSelection
    ais = AIS_ColoredShape(obj.wrapped)
TypeError: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. OCP.AIS.AIS_ColoredShape(theShape: OCP.TopoDS.TopoDS_Shape)
    2. OCP.AIS.AIS_ColoredShape(theShape: OCP.AIS.AIS_Shape)

Invoked with: <OCP.gp.gp_Vec object at 0x7fe9b8bf1370>
```

lorenzncode, 2021-08-21:
>Isn't it the other way around?

Oops, yes you are right, I got that backwards!

The bottom item has no parent (where you started the part).  The top item has the most parents (where you finished).

>Wouldn't it be a good idea to have it hide automatically when you select an "older" item? And when you select the last item in the list, which is the final object, it could unhide back.

Maybe as some other future mode of inspection but I think the user should be allowed to display the part along with object inspection at the same time.  Currently the user has full control over the parts to display.  It can be useful to display the part along with object inspection at the same time in some cases.

>Do you want me to create a separate issue for this?

@adam-urbanczyk  can decide.

>When an intermediate workplane is displayed, how do I know what are its x, y and z axis oriented?

CQ-Editor displays a grid.  Example:

```
r = cq.Workplane("XY").rect(10,10).extrude(5).faces(">Y").workplane().center(3,3).circle(1).cutBlind(-2)
```

[image]

The Introspective Example shows how to access/inspect a Workplane's plane attribute (Plane object) which might help as well?

lorenzncode, 2021-08-21:
>I found another error:
Yes, can reproduce with say the following on inspecting the first Workplane in the chain (which has Vector in objects list).
```
import cadquery as cq
r = cq.Workplane().pushPoints([(1,2)]).circle(1).extrude(1)
```