CadQuery/CQ-editor #282: Incorrect default orientation

revolter, 2021-08-14:
I think that the default orientation should have the `x` axis oriented towards the bottom left corner:

[image]

instead of the bottom right corner:

[image]

adam-urbanczyk, 2021-08-14:
Why is the current orientation incorrect? I find it more intuitive.

revolter, 2021-08-16:
Nevermind. I assumed that it should be this way because I have no experience with working with 3D softwares. I now head that the second default orientation is used both in FreeCAD and Unity.

closed 2021-08-16 by revolter