CadQuery/CQ-editor #283: Fit automatically option is ignored

revolter, 2021-08-14:
If I disable the "fit automatically" option, and use the auto reload option to edit the script in Sublime Text, the object still gets auto fitted.

adam-urbanczyk, 2021-08-14:
I cannot reproduce this.

revolter, 2021-08-17:
I can't reproduce it either. I think that it happened in the same app session in which I disabled this option, and after at least one app restart, it works. But I'm now sure.

jmwright, 2021-08-17:
In my experience some of the settings do require an app restart.

revolter, 2021-08-17:
Could we identify the ones that need it, and warn the users about it when they change them?

jmwright, 2021-08-17:
How would the user be warned? On some apps the label of a setting will say "(Requires Restart)", and sometimes there's a popup dialog that tells the user that the setting they just changed requires a restart.

revolter, 2021-08-17:
Yeah, one of those two. But if you think that it's not necessary, obviously, feel free to close this issue.

closed 2025-03-09 by jmwright