CadQuery/CQ-editor #288: Development package fails in ArchLinux

mantielero, 2021-09-07:
I am trying to use the [development package](https://github.com/CadQuery/CQ-editor#development-packages) under ArchLinux, but I am failing to make it work.

I am getting the error:
```
FileNotFoundError: [Errno 2] No such file or directory: '/home/jose/bin/CQ-editor/debugpy/_vendored'
[42577] Failed to execute script run
```

Any clue of what am I doing wrong?

adam-urbanczyk, 2021-09-07:
Ask the maintainer of the package. It is not provided by the CQ team.

mantielero, 2021-09-07:
I am not talking about any ArchLinux packages, but the development packages provided [here](https://github.com/jmwright/CQ-editor/actions/runs/1204158454) in Github under Artifacts.

I run the *CQ-editor.sh* script and it stops with that error.
```
$ ./CQ-editor.sh 
Traceback (most recent call last):
  File "run.py", line 14, in <module>
  File "cq_editor/__main__.py", line 16, in main
  File "cq_editor/main_window.py", line 38, in __init__
  File "cq_editor/main_window.py", line 93, in prepare_panes
  File "cq_editor/widgets/console.py", line 20, in __init__
  File "site-packages/traitlets/traitlets.py", line 1032, in __new__
  File "site-packages/traitlets/traitlets.py", line 1068, in setup_instance
  File "site-packages/traitlets/traitlets.py", line 1055, in setup_instance
  File "site-packages/traitlets/traitlets.py", line 1863, in instance_init
  File "site-packages/traitlets/traitlets.py", line 1868, in _resolve_classes
  File "site-packages/traitlets/traitlets.py", line 1694, in _resolve_string
  File "site-packages/traitlets/utils/importstring.py", line 30, in import_item
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "/usr/share/miniconda3/envs/test/lib/python3.8/site-packages/PyInstaller/loader/pyimod03_importers.py", line 623, in exec_module
  File "site-packages/ipykernel/inprocess/ipkernel.py", line 13, in <module>
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "/usr/share/miniconda3/envs/test/lib/python3.8/site-packages/PyInstaller/loader/pyimod03_importers.py", line 623, in exec_module
  File "site-packages/ipykernel/ipkernel.py", line 20, in <module>
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "/usr/share/miniconda3/envs/test/lib/python3.8/site-packages/PyInstaller/loader/pyimod03_importers.py", line 623, in exec_module
  File "site-packages/ipykernel/debugger.py", line 17, in <module>
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "/usr/share/miniconda3/envs/test/lib/python3.8/site-packages/PyInstaller/loader/pyimod03_importers.py", line 623, in exec_module
  File "site-packages/debugpy/server/__init__.py", line 9, in <module>
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "/usr/share/miniconda3/envs/test/lib/python3.8/site-packages/PyInstaller/loader/pyimod03_importers.py", line 623, in exec_module
  File "site-packages/debugpy/_vendored/__init__.py", line 18, in <module>
FileNotFoundError: [Errno 2] No such file or directory: '/home/jose/bin/CQ-editor/debugpy/_vendored'
[51533] Failed to execute script run
```

jmwright, 2021-09-08:
@mantielero Does the stable release package work for you? https://github.com/CadQuery/CQ-editor/releases/tag/0.2

mantielero, 2021-09-08:
Yes. That works.

jmwright, 2021-09-08:
@mantielero Something changed in the very latest build, and it doesn't work in Ubuntu 18.04 either. Try the build just before that. It works for me. https://github.com/jmwright/CQ-editor/actions/runs/1180360559

mantielero, 2021-09-08:
Yes you are right. That works for me too.

jmwright, 2021-09-08:
@mantielero I added the needed dependencies and it's working again in Ubuntu 18.04. Please try build 85 https://github.com/jmwright/CQ-editor/actions/runs/1213505147

mantielero, 2021-09-08:
This build works too.

closed 2021-09-08 by mantielero