CadQuery/CQ-editor #289: Feature Request - AppImage creation

mantielero, 2021-09-08:
I think it would be nice providing an AppImage for Linux. I think AppImage suits better than flatpak/snap for CQ-editor. And it is easier to create it (this is my first AppImage; I just leart how to do it today).

The process goes as follows (more or less):
1. Create the folder `CQ-editor.AppDir`
2. Unzip the content of one of the .zip files (stable or development versions) in that folder.
3. Rename the entry point file:
```sh
mv CQ-editor.sh AppRun
```
4. Modify the content of `AppRun`:
```sh
#!/bin/sh
export QT_MAC_WANTS_LAYER=1

HERE="$(dirname "$(readlink -f "${0}")")"

#chmod u+x ./CQ-editor/CQ-editor
#./CQ-editor/CQ-editor
#chmod u+x "$HERE/CQ-editor/CQ-editor"
exec "$HERE/CQ-editor/CQ-editor"
```
5. Get the logo:
```
$ wget https://raw.githubusercontent.com/CadQuery/CQ-editor/master/icons/cadquery_logo_dark.svg
$ mv cadquery_logo_dark.svg CQ-editor.svg
$ cp CQ-editor.png .DirIcon
```
6. Create the file `CQ-editor.desktop`:
```sh
[Desktop Entry]
Type=Application
Terminal=false
Icon=CQ-editor
Exec=AppRun
Name=CQ-editor
GenericName=Cadquery Editor
Categories=Graphics;
Comment=CadQuery GUI editor based on PyQT
```
7. Create the appimage file:
```sh
$ appimagetool CQ-editor.AppDir/
```
(this will create a file like: `CQ-editor-x86_64.AppImage`) 

------

The benefits as a user:
- You download the file and just chmod it to make it executable. You don't need to unzip it. You don't have thousands of files in a folder structure.
- You can easily integrate it in the desktop by using things like [AppImageLauncher](https://github.com/TheAssassin/AppImageLauncher). Just doing:
  ```sh
  ail-cli integrate CQ-editor-x86_64.AppImage
  ```
  and you get your icon in Gnome. 
- Several versions can coexist easily with proper names

In theory this can be done automatically with travis-ci. But I have never used it.

adam-urbanczyk, 2021-09-08:
PRs are welcome - preferably as an azure pipeline.

mantielero, 2021-09-18:
I don't know how to do that (I have never done CI before). 

I have been playing with Azure (run the basic tutorial), I forked the repository and I have tried to add a new job just to know what should I modify by adding the following in `azure-pipelines.yml`:
```yaml
- job: Work
  steps:
  - script: echo Hello, world!
    displayName: 'Run a one-line script'

  # This malformed command causes the job to fail
  # Only run this command if the succeed variable is set to false
  - script: git clone malformed input
    condition: eq(${{ parameters.succeed }}, false)
```

This fails complaining about:
```
Repository templates references endpoint CadQuery which does not exists or is not authorize for use
```

How should I proceed?