CadQuery/CQ-editor #298: Fails to start on Arch Linux

flaviut, 2021-11-03:
When I run the pre-packaged binary on Arch Linux, I get the following error in the terminal:

```
Namespace(filename=None)
libGL error: MESA-LOADER: failed to open radeonsi: /home/user/tmp/CQ-editor-Linux-x86_64/CQ-editor/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/dri/radeonsi_dri.so) (search paths /usr/lib/dri, suffix _dri)
libGL error: failed to load driver: radeonsi
libGL error: MESA-LOADER: failed to open radeonsi: /home/user/tmp/CQ-editor-Linux-x86_64/CQ-editor/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/dri/radeonsi_dri.so) (search paths /usr/lib/dri, suffix _dri)
libGL error: failed to load driver: radeonsi
libGL error: MESA-LOADER: failed to open swrast: /home/user/tmp/CQ-editor-Linux-x86_64/CQ-editor/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/dri/swrast_dri.so) (search paths /usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
libGL error: MESA-LOADER: failed to open radeonsi: /home/user/tmp/CQ-editor-Linux-x86_64/CQ-editor/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/dri/radeonsi_dri.so) (search paths /usr/lib/dri, suffix _dri)
libGL error: failed to load driver: radeonsi
libGL error: MESA-LOADER: failed to open radeonsi: /home/user/tmp/CQ-editor-Linux-x86_64/CQ-editor/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/dri/radeonsi_dri.so) (search paths /usr/lib/dri, suffix _dri)
libGL error: failed to load driver: radeonsi
libGL error: MESA-LOADER: failed to open swrast: /home/user/tmp/CQ-editor-Linux-x86_64/CQ-editor/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/dri/swrast_dri.so) (search paths /usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
```

flaviut, 2021-11-03:
This is because the package uses an old libstdc++ that my video driver binary is not compatible with. A possible solution may be to not ship libraries that have a strong ABI policy like libstdc++ with the pre-packaged binary.

A workaround is to delete the library: `rm CQ-editor/libstdc++.so.6`.

adam-urbanczyk, 2021-11-06:
Thanks for reporting and finding the solution @flaviut . I'll reopen to not forget about this.

lorenzncode, 2022-01-09:
I can reproduce this with Fedora 35 and CQ-editor version 0.2.

The CQ-editor GUI appears but render fails.  The Log viewer then shows:
```
[2022-01-09 16:46:07.983459] ERROR: CQ GUI: Uncaught exception occurred
Traceback (most recent call last):
  File "cq_editor/widgets/occt_widget.py", line 131, in paintEvent
  File "cq_editor/widgets/occt_widget.py", line 153, in _initialize
OCP.Aspect.Aspect_GraphicDeviceDefinitionError: OpenGl_Window::CreateWindow: glXCreateContext failed.
```

Both the rm solution mentioned above or LD_PRELOAD (as mentioned by an ubuntu 21.10 user in google groups) workaround the issue:
- `rm ./CQ-editor-0.2/libstdc++.so.6`
- `LD_PRELOAD=/lib64/libstdc++.so.6`

jmwright, 2022-01-12:
The development builds of CQ-editor have been fixed and no longer bundle libstdc++ in the Linux package. It would be great if it received some testing by varied users.

https://github.com/jmwright/CQ-editor/actions/workflows/pyinstaller-builds-actions.yml

lorenzncode, 2022-01-13:
I tested build 117.  I was able to install/run on:
- Ubuntu 20.04
- Ubuntu 21.10
- Fedora 35 (had to create a symlink to libblas64.so.3)

To resolve ImportErrors on Ubuntu:
`$ sudo apt install libglew2.1`
`$ sudo apt install libblas3`

To resolve Import Errors on Fedora:
`$ sudo dnf install libGLEW`
create libblas.so.3 symlink

adam-urbanczyk, 2022-01-13:
Shouldn't we package blas?

jmwright, 2022-01-15:
> Shouldn't we package blas?

Based on this conversation and conversation on Discord, yes. I just need to figure out how to force PyInstaller to bundle it.

jmwright, 2022-03-09:
@lorenzncode What did the libblas symlink you created point from/to? The CQ-editor package now includes libblas3, but CQ-editor still won't start on the Fedora 35 VM that I have setup for this.

lorenzncode, 2022-03-09:
@jmwright I still have these test builds around - both 117 and 123.

For 117, yes I had created a symlink before I realized I should simply do ```sudo dnf install blas``` (I already had blas64 installed). 

To answer your question, this is the symlink, but would not recommend - better to install blas.:
```
[lorenzn@fedora CQ-editor]$ uname -a
Linux fedora 5.16.11-200.fc35.x86_64 #1 SMP PREEMPT Wed Feb 23 17:08:49 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
[lorenzn@fedora CQ-editor]$ pwd
/home/lorenzn/cq-editor-build117/CQ-editor
[lorenzn@fedora CQ-editor]$ ls -l libblas*
lrwxrwxrwx. 1 lorenzn lorenzn 21 Jan 12 21:08 libblas.so.3 -> /lib64/libblas64.so.3
[lorenzn@fedora CQ-editor]$ rpm -qf /lib64/libblas64.so.3.10.0
blas64-3.10.0-4.fc35.x86_64
```

For 123, (different VM), again this time did ```sudo dnf install blas```.  123 did not require system libGLEW.
```
[lorenzn@fedora CQ-editor]$ uname -a
Linux fedora 5.14.10-300.fc35.x86_64 #1 SMP Thu Oct 7 20:48:44 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
[lorenzn@fedora CQ-editor]$ pwd
/home/lorenzn/cq-editor-build123/CQ-editor
[lorenzn@fedora CQ-editor]$ ls -l libblas*
ls: cannot access 'libblas*': No such file or directory
[lorenzn@fedora CQ-editor]$ ls -l /lib64/libblas.so.3
lrwxrwxrwx. 1 root root 17 Dec  8 16:46 /lib64/libblas.so.3 -> libblas.so.3.10.0
[lorenzn@fedora CQ-editor]$ rpm -qf /lib64/libblas.so.3
blas-3.10.0-4.fc35.x86_64
```

Anyway if the pyinstaller version packages blas, the system blas should not be required right?

jmwright, 2022-03-10:
> Anyway if the pyinstaller version packages blas, the system blas should not be required right?

Something is still missing. I can see libblas in the package, but since you're still having to install the blas package we're missing dependencies somewhere.

roipoussiere, 2022-03-23:
Not sure if this is related to this issue, but using CQE 0.2.0 on Kubuntu 21.10, I still have the error `OCP.Aspect.Aspect_GraphicDeviceDefinitionError: OpenGl_Window::CreateWindow: glXCreateContext failed.`, which I fixed by replacing the last line on the launcher with `LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6 ./CQ-editor/CQ-editor` as explained in [this GoogleGroups topic](https://groups.google.com/g/cadquery/c/-MQily471Mg).

Should I create a new issue for this?

jmwright, 2022-03-23:
> Should I create a new issue for this?

You can create one, but the change will most likely be made on my fork. That startup script should really search for libstdc++.so.6 on the local system and use that path instead of using a hard-coded one. I think the path can vary on another distros.

jmwright, 2022-04-14:
The [latest development build](https://github.com/jmwright/CQ-editor/actions/runs/2169082713) runs correctly on a fresh Manjaro Arch XFCE VM without installing anything. Feel free to reopen if there are other issues on Arch.

closed 2022-04-14 by jmwright

jdegenstein, 2022-10-14:
According to the pyinstaller devs, they recommend that you do NOT remove e.g. libstdc++.so.6 because it will irreparably break older linux distros that may not have a new enough version of the std c++ libraries. 

See here: https://github.com/pyinstaller/pyinstaller/issues/6993

Of course that still leaves us with an issue where the bundled libstdc++ library has some incompatibility with the packages on newer systems. The solution discussed so far would be to include a version check of the static libstdc++ versus the system libstdc++, and presumably the newer should be used (big assumption, not confirmed). This issue here describes a way to parse the library for included versions:

See https://github.com/Chia-Network/chia-blockchain/issues/13242
```
$strings /usr/lib/x86_64-linux-gnu/libstdc++.so.6 | grep GLIBCXX_3.4.26
GLIBCXX_3.4.26
```
The above returns a match when the version is found. There must be a shell script lying around that solves this problem. I believe using `LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6 ./CQ-editor/CQ-editor` on the newer systems will be a good workaround once the logic is complete.

jmwright, 2022-10-17:
Here is a new static build with libstdc++ still bundled if anyone wants to try it.

https://github.com/jmwright/CQ-editor/actions/runs/3259613214