CadQuery/CQ-editor #301: Watch imported modules feature doesn't work consistently

possibilities, 2021-11-26:
Seems that the "watch imported modules" feature breaks after an error occurs (or maybe it's more specific than that?):

[image]

One thing to note about this video is that the model I'm rendering is very inefficient (my fault) so after each time I click the button to refresh I put my mouse on the viewport and try to reposition just to show that either nothing is happening (I can move the viewport) or something is happening (I can't move the viewport while it works):

https://youtu.be/ExZ4Mp4Tu4M

I might try to solve this myself but it's my first time really getting in the codebase and not much experience with python so feel free to work in parallel or check with me on progress.

adam-urbanczyk, 2021-11-26:
Should be solved by #300 . Could you pull and check?

possibilities, 2021-11-26:
I couldn't figure out how to install or run from the repo initially... but now that it's merged should I be able to try it by running this again?

```
conda install -c cadquery -c conda-forge cq-editor=master --yes
```

Otherwise can you explain the best way to run it from the repo? Thanks!

possibilities, 2021-11-26:
Seems fixed, thanks @lorenzncode and @adam-urbanczyk !

closed 2021-11-26 by possibilities