CadQuery/CQ-editor #303: How to get running filename

threed-factory-store, 2021-12-07:
Sorry, I couldn't find a community forum so I'm asking here:

I have a CadQuery script that I'm running in cq-editor.  How can I get the filename of my script?

lorenzncode, 2021-12-07:
The name of the opened file is displayed in the window title.

Example where I opened "myscript.py":
[image]

Also see the CadQuery GitHub page _Getting help_ section.  There are links to the Google Group and Discord channel there.

adam-urbanczyk, 2021-12-08:
If the question was about getting it in the script itself - there is no intentionally exposed API to do that.