CadQuery/CQ-editor #304: Change color scheme of widgets

machine85, 2021-12-08:
Hello,
i download the 0.2 release,
i manage to change the color scheme of the Editor widget by Edit-->Preferences-->Code Editor.
how to change the color of the other widgets: Console, Curren traceback, Log viewer.. ?

Regards
Tom

adam-urbanczyk, 2021-12-08:
Not implemented ATM.

axhan, 2022-04-08:
Not 100% what the OP desired, but I just implemented selectable dark+light themes in this fork: https://github.com/axhan/CQ-editor_themed.

machine85, 2022-04-18:
Thank you very much sir for your great effort.
if i already install cq-editor in cinda by the command:
*conda install -c cadquery -c conda-forge cq-editor=master*

*does i can add your themed to the existing install ?*
*or the only way is to install your version of cq-editor *

*best regards,*
*Tomer*

‫בתאריך יום ו׳, 8 באפר׳ 2022 ב-22:03 מאת ‪Anders Vedmar‬‏ <‪
***@***.***‬‏>:‬

machine85, 2022-04-19:
Hello,
i managed to add the theme,
1. i got into the theme folder and copy the folder "cq_editor"
2. i paste the folder instead of the original folder of the modul:
*miniconda/envs/cadquery-master/lib/python3.9/site-packages/cq_editor*

i add a image of the theme
thank you very much !!

‫בתאריך יום ב׳, 18 באפר׳ 2022 ב-21:18 מאת ‪tomer u‬‏ ***@***.***
‬‏>:‬

inktrap, 2025-02-17:
So what is needed to add the changes from @axhan ?

Given that their fork is many commits behind now, I added very basic dark theme support myself, just used the simple example from https://github.com/ColinDuquesnoy/QDarkStyleSheet and it works, see https://github.com/inktrap/CQ-editor and I didn't open a PR because some more changes are necessary.

It would be easy to just call `app.setStyleSheet(qdarkstyle.load_stylesheet())` for every app object:

```
> rg "app = QApplication"
cq_editor/__main__.py
13:app = QApplication(sys.argv, applicationName=NAME)

cq_editor/widgets/viewer.py
454:    app = QApplication(sys.argv)

cq_editor/widgets/editor.py
317:    app = QApplication(sys.argv)

cq_editor/widgets/console.py
75:    app = QApplication(sys.argv)
```

If there is no progress for this issue (no blame!) you can clone the repo yourself and install it with something like `uv tool  install -e ~/CQ-editor` and run it with `uv tool run CQ-editor`. Cleaner than to change something in some virtualenv directly. I kept the changes to a minimum so merges should just work.

closed 2025-02-28 by jmwright