CadQuery/CQ-editor #305: Can't get objects to intersect.

abalter, 2022-01-03:
I'm trying to create to tori (dounts) that intersect. The two created by the code below just touch. if I try to change the `moveTo` command for `result2` to something like `moveto(24, 12)` so they would intersect, I get

>Standard_NullObject: BRep_Tool: TopoDS_Vertex hasn't gp_Png

Whatever that means.

```
import cadquery as cq

result1 = (
    cq
    .Workplane()
    .moveTo(24,0)
    .circle(8)
    .revolve()
    )

result2 = (
    cq
    .Workplane()
    .moveTo(24,16)
    .circle(8)
    .revolve()
    )

result = result1.union(result2)

show_object(result)
```

jmwright, 2022-01-04:
The CAD kernel is probably doing something weird. If you disable clean, it will work.
```python
result1 = (
    cq
    .Workplane()
    .moveTo(24,0)
    .circle(8)
    .revolve()
    )

result2 = (
    cq
    .Workplane()
    .moveTo(24,12)
    .circle(8)
    .revolve()
    )

result = result1.union(result2, clean=False)

show_object(result)
```
You can also fix it be using `glue=True` in `union` since the donuts are overlapping.

adam-urbanczyk, 2022-01-04:
Thanks for reporting, but you are posting to a wrong tracker. This issue is not CQ-editor specific.

closed 2022-01-04 by adam-urbanczyk

abalter, 2022-01-05:
Thanks @jmwright!