CadQuery/CQ-editor #306: Latest release not compatible with latest docs

qwelyt, 2022-01-04:
Trying to move from OpenSCAD to cad-query. Got the latest release zip as that was the easiest way to install cq. However, trying to follow the code written in the latest documentation is not possible due to cq-editor not supporting Sketch. I saw that a support for this was merged in some time ago.

Is it possible to create a new release so people who are trying out CQ for the first time can follow the docs with the editor?

jmwright, 2022-01-04:
There is a PyInstaller problem with the latest development builds of CQ-editor that has caused the zip files to fall out of date temporarily. I'm working on fixing that, but the problem is not a simple one. In the meantime, you could go ahead and try the Anaconda installation to get going with the latest version.

jmwright, 2022-01-12:
The development builds of CQ-editor have been fixed. @qwelyt Please try the latest build since it should be up-to-date with the latest CadQuery features. Click the latest "build" with a green checkmark [here](https://github.com/jmwright/CQ-editor/actions/workflows/pyinstaller-builds-actions.yml), scroll down to the Artifacts section, and download the package for your OS.

NothanUmber, 2022-01-16:
Thanks for getting the installer running again - think having an easily downloadable package like for OpenSCAD is very helpful to get it in the hands of more people!
Tried the Windows build 125, couldn't get it to render anything. Getting this error message:
[2022-01-16 13:18:13.732858] ERROR: CQ-Editor: Uncaught exception occurred
Traceback (most recent call last):
  File "cq_editor\widgets\occt_widget.py", line 131, in paintEvent
  File "cq_editor\widgets\occt_widget.py", line 153, in _initialize
  File "cq_editor\widgets\occt_widget.py", line 159, in _get_window_win
ModuleNotFoundError: No module named 'OCP.WNT'; 'OCP' is not a package

So OCP doesn't seem to be packaged yet?

adam-urbanczyk, 2022-01-17:
Should be solved by: https://github.com/CadQuery/OCP/pull/75/

jmwright, 2022-01-17:
[Development build 127](https://github.com/jmwright/CQ-editor/actions/runs/1709533939) should include the fix mentioned by @adam-urbanczyk above. @NothanUmber can you please try it out to see it it's resolved?

NothanUmber, 2022-01-17:
@jmwright Still getting exactly the same error message with build 127.

NothanUmber, 2022-01-17:
When comparing my cq-dev conda based cadquery version with the bundled one then one difference is that the bundled one comes with OCP.cp38-win_amd64.pyd while my (working) version comes with OCP.cp39-win_amd64.pyd.
Just naively copying over the OCP.cp39* file didn't solve the issue though.
Perhaps the bundled Python interpreter is incompatible with the bundled OCP?

adam-urbanczyk, 2022-01-18:
Build 127 still bundles OCP 7.5.3.0 build 2 and it should be build 3 (see https://github.com/jmwright/CQ-editor/runs/4845131285?check_suite_focus=true#step:4:154 ). I'm going to mark build 2 as broken, maybe it'll help.

jmwright, 2022-01-19:
@NothanUmber The OCP Python version has to match the embedded version of Python, which is pinned to Python 3.8 in the development packages for now. That's why manually copying the OCP lib for Python 3.9 didn't work. Hopefully the next run will pick up the correct OCP build.

@adam-urbanczyk Thanks for the heads-up. I've triggered another build of the packages, and when it's done I'll check to make sure it's grabbing build 3 of OCP 7.5.3.

jmwright, 2022-01-19:
Run 128 seems to correctly pull OCP 7.5.3 build 3, and I have confirmed that it runs and I can render a CadQuery model in Windows.

https://github.com/jmwright/CQ-editor/actions/runs/1718304321

NothanUmber, 2022-01-25:
Just tried build 129 - works! Thank you!

jpoles1, 2022-02-06:
This fixed it for me as well, but was very confused for a little bit. This should probably be posted to the readme or dropped into a new release!

GandalfDG, 2022-04-20:
Is there an update on this? I installed the latest release on my Windows machine, and it still doesn't appear to ship with the latest version of CadQuery that supports the Sketch API. I have worked around it for my installation, but will there be a new release to address this in a more permanent way? Going to the build linked above is no longer an option as the build artifacts have expired.

GandalfDG, 2022-04-20:
I will try it out. I was mainly wondering whether there would be a new release of CQ-editor on the github page so that the action new users would default to for installing CQ-editor would give them a version which works with the latest docs out of the box.

jpoles1, 2022-08-12:
I think that this is still an issue! If new users land on the cadquery github page, they're likely going to go first to releases and download a version that is out of date. 

@jmwright, is there any way to "release" a version that is aligned with the current CadQuery version (and documentation)? I think it'd help new users get started with CadQuery much more easily. 

I'm working on a cadquery cheatsheet, and debating whether I should include instructions on how to find the latest CQ-Editor version under the workflow/action tab with the green checkmarks; it'd be ideal to just link to "releases".

jmwright, 2022-08-15:
@jpoles1 

> is there any way to "release" a version that is aligned with the current CadQuery version (and documentation)?

https://github.com/CadQuery/CQ-editor/issues/274

natewaddoups, 2022-08-27:
+1 for this issue. I liked what I heard about cq-editor so I did the obvious thing: came to the cq-editor github project and installed the latest release. 

And then I tried an example from the documentation and got an error message instead of geometry.

One can only wonder how many potential users cadquery has lost to the "module cadquery has no attribute Sketch" error.

bschwind, 2023-07-12:
Yup, I ran into this too. Read the README, went to the "latest release" link, downloaded it, and immediately ran into issues with the `Sketch` API missing. 

Of course I eventually came across this issue and others, and was able to download a more recent build. I would definitely consider this a huge stumbling block for beginners though.