CadQuery/CQ-editor #309: No module named 'OCP.WNT'; 'OCP' is not a package

jschoch, 2022-01-16:
i'm trying to run cq-editor along with cadquery master and cq_gears.

```python
[2022-01-16 19:57:37.016681] ERROR: CQ-Editor: Uncaught exception occurred
Traceback (most recent call last):
  File "C:\Users\jesse\anaconda3\envs\py38\lib\site-packages\cq_editor\widgets\occt_widget.py", line 131, in paintEvent
    self._initialize()
  File "C:\Users\jesse\anaconda3\envs\py38\lib\site-packages\cq_editor\widgets\occt_widget.py", line 153, in _initialize
    self.view.SetWindow(wins.get(platform,self._get_window_linux)(self.winId()))
  File "C:\Users\jesse\anaconda3\envs\py38\lib\site-packages\cq_editor\widgets\occt_widget.py", line 159, in _get_window_win
    from OCP.WNT import WNT_Window
ModuleNotFoundError: No module named 'OCP.WNT'; 'OCP' is not a package

```

I'm a python newbie.  Building with conda took an hour...

command to install 

`conda create -n py38 python=3.8 anaconda cadquery=master cq-editor=master ocp -c cadquery -c conda-forge`

I got cq_gears working in jupyter, no idea why the issue with OCP.

jmwright, 2022-01-16:
Similar problem was posted recently [here](https://github.com/CadQuery/CQ-editor/issues/306#issuecomment-1013875430)

adam-urbanczyk, 2022-01-17:
Should be solved by this PR: https://github.com/CadQuery/OCP/pull/75/

jschoch, 2022-01-17:
thanks!  

sorry if this is a newb question but: how do I get the new version installed via conda?  I'm on win 10.  There doesn't seem to be a @master.  Is the azure build pipeline making releases?

jmwright, 2022-01-17:
[Development build 127](https://github.com/jmwright/CQ-editor/actions/runs/1709533939) should include the fix Adam mentioned above.

jmwright, 2022-01-17:
@jschoch If you follow the conda instructions in the [getting started](https://github.com/CadQuery/cadquery#getting-started) section of the CadQuery readme, you should be getting the master branch.

jschoch, 2022-01-17:
i ran
`conda create -n py38_2 python=3.8 anaconda cadquery=master cq-editor=master ocp -c cadquery -c conda-forge`

then i ran 

` conda update -n py38_2 ocp -c cadquery`

finally

`pip install git+https://github.com/meadiode/cq_gears.git@main`

and cq_gears is working!!!!

Thanks for the help!!!

closed 2022-01-17 by jschoch