CadQuery/CQ-editor #310: cq-editor window does not show on macOS Monterey version 12.1

etijskens, 2022-01-21:
Installed CadQuery and cq-editor as prescribed. 
  > conda activate base 
  > cq-editor 
  Namespace(filename=None)
Nothing further happens.

jmwright, 2022-01-21:
I tested on Catalina 2 days ago and CQ-editor ran fine, so it must be a difference between MacOS versions and not CQ-editor itself. Also, if you're trying to run it on an M1 Mac, you probably won't have much luck.

You might also try setting the environment variable specified [here](https://github.com/jmwright/CQ-editor/blob/master/pyinstaller/CQ-editor.sh#L2) before running CQ-editor.

raquo, 2022-01-27:
I have a similar looking issue that started yesterday, on an intel mac on macos catalina 10.15.7. Strangely, it was working fine until I rebooted. I'm not 100% sure, but I don't think any system updates were applied in the process.

@etijskens When you run `cq-editor` on your mac, does the command keep running, or does it quit and return you to the command prompt? For me it quits after a second or so.

For the record, I tried the env variable too, didn't seem to do anything:

```
(base) ➜  ~ export QT_MAC_WANTS_LAYER=1
(base) ➜  ~ cq-editor
(base) ➜  ~ 
```

etijskens, 2022-01-27:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

It keeps running. It doesn’t return at all.

Kindest regards,

Dr [Engel]bert Tijskens - HPC Analyst/Consultant
Flemish Supercomputer Center
University of Antwerp / Computational mathematics
++32 3 265 3879, ++32 494 664408

closed 2025-02-11 by jmwright