CadQuery/CQ-editor #312: cq-editor exits without message on import of namespace package

lorenzncode, 2022-01-27:
To reproduce, import a namespace package.  cq-editor fails to launch and exits without any message.

trace
```
  File "/home/lorenzn/tools/miniforge/envs/cq753/lib/python3.9/modulefinder.py", line 77, in _find_module
    if spec.loader.is_package(name):
AttributeError: 'NoneType' object has no attribute 'is_package'
```

Also see: https://bugs.python.org/issue40350

adam-urbanczyk, 2022-01-28:
That looks like a upstream bug (of python), or am I missing something?

lorenzncode, 2022-01-29:
Yes, there is a related upstream modulefinder bug.  I'd propose to handle the uncaught exception raised by `modulefinder.run_script`.  Currently only `SyntaxError` is handled but now I see that other exceptions can be raised when finding the imported modules.

Here is an example similar to #99.
```
$ tree -L 2
.
├── main.py
└── mybox
    └── driving_dims.py

1 directory, 2 files
```

main.py:
```python
import cadquery as cq
import mybox.driving_dims as driving_dims

a = cq.Workplane("front").box(*driving_dims.box)
cq.exporters.export(a, "a.step")
```

mybox/driving_dims.py:
```python
box = (100, 20, 30)

```

There is no problem with CQ as a library.:
```
$ python main.py

*******************************************************************
******        Statistics on Transfer (Write)                 ******

*******************************************************************
******        Transfer Mode = 0  I.E.  As Is       ******
******        Transferring Shape, ShapeType = 0                      ******
** WorkSession : Sending all data
 Step File Name : a.step(350 ents)  Write  Done
```

CQ-Editor does not open.:
```
$ cq-editor main.py 
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Namespace(filename='main.py')
```

If the script had previously been saved to the QSettings file, `$ cq-editor` without a script as argument also fails.

PR #314 (2022-02-01): Handle exceptions when determining imported modules (#312)
This is to resolve #312 where CQ-Editor can get into a state where it does not launch and no error message is issued.
closed 2022-02-01 by adam-urbanczyk