CadQuery/CQ-editor #313: define __name__ as '__main__' in CQEditor?

NothanUmber, 2022-01-27:
Assume you have an assembly.py than includes several other scripts part1.py part2.py etc. and assembles those.
Now it would be nice if I could still open the individual partX.py files in CQEditor to just see that particular part. But also be able to load assembly.py to see everything.

My assumption how to do this was to have section like this in every part file as well as the assembly:
if __name__ == '__main__':
    result = main()
    show_object(result)

So main() show_object() are only called for a part if I directly load that part file - but not if I open the assembly file.

Currently the __name__ mechanism (that sets this to __main__ in the main script but not the imported packages) doesn't seem to be supported in CQEditor yet (when guarding show_object with that condition then it never gets executed).

Would it make sense to add support for that? Or is there another way to reach the same goal?

voneiden, 2022-02-09:
`show_object` is only defined in the main module that you have open in CQ-Editor. So in your part file you could for example  call show_object conditionally

```
if 'show_object' in locals():
    show_object(...)
```

NothanUmber, 2022-02-09:
Thanks @voneiden, good idea! Didn't have on the radar that show_object is only defined in the main module (and not in included modules) - this should solve this then!

closed 2022-02-22 by NothanUmber