CadQuery/CQ-editor #325: Error on startup: ModuleNotFoundError: No module named 'zmq.backend.cython'

roipoussiere, 2022-03-23:
Using [the last dev build](https://github.com/jmwright/CQ-editor/actions/runs/2013452468), I get this error when I try to run CQ-editor:

```
Traceback (most recent call last):
  File "run.py", line 12, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "cq_editor/__main__.py", line 12, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "cq_editor/main_window.py", line 9, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "cq_editor/widgets/console.py", line 5, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "qtconsole/inprocess.py", line 5, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "ipykernel/__init__.py", line 2, in <module>
    # -----------------------------------------------------------------------------
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "ipykernel/connect.py", line 10, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "jupyter_client/__init__.py", line 6, in <module>
    # (see LICENSE.txt in this directory and NOTICE.txt in the root for details)
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "jupyter_client/asynchronous/__init__.py", line 1, in <module>
    # -*- coding: utf-8 -*-
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "jupyter_client/asynchronous/client.py", line 6, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "jupyter_client/channels.py", line 12, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "zmq/__init__.py", line 103, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "zmq/backend/__init__.py", line 32, in <module>
  File "zmq/backend/__init__.py", line 27, in <module>
  File "zmq/backend/select.py", line 32, in select_backend
  File "importlib/__init__.py", line 127, in import_module
ModuleNotFoundError: No module named 'zmq.backend.cython'
[4117] Failed to execute script 'run' due to unhandled exception!
```

Using [this one](https://github.com/jmwright/CQ-editor/actions/runs/1685548993) published 2 months ago, it works as expected.

I'm on Kubuntu 21.10.

jmwright, 2022-03-25:
I have seen this problem during testing on Fedora as well. Can either of you confirm if the same error happens when using conda?

roipoussiere, 2022-03-25:
I tried to execute `conda install -c cadquery -c conda-forge cq-editor=master` on a miniconda docker container but my PC freezes before it's finished.

My PC has some troubleshooting and regularly freezes when a huge memory load is needed, so it's not an error on the cqe side, but unfortunately I just can't help here.

pauljurczak, 2022-04-05:
Same problem here as I mentioned in duplicate #333. Conda install was successful, though. 

I also tried build 130, and it seems to be working fine, but outputs this:
```
paul@desktop:~/apps/cq-editor-0.3.130$ ./CQ-editor.sh
This version of python seems to be incorrectly compiled
(internal generated filenames are not absolute).
This may make the debugger miss breakpoints.
Related bug: http://bugs.python.org/issue1666807
Namespace(filename=None)
```

jmwright, 2022-04-14:
Multiple unrelated problems hit at once that made it time consuming to fix this one, but I think it's working again. Please try [this development build](https://github.com/jmwright/CQ-editor/actions/runs/2169082713) and let me know how you get along.

pauljurczak, 2022-04-15:
It loads and seems to be working fine on Ubuntu 20.04. Here is the output:
```
paul@desktop:~/apps/cq-155$ ./cq.sh
0.00s - Debugger warning: The os.path.realpath.__code__.co_filename (posixpath.py)
0.00s - is not absolute, which may make the debugger miss breakpoints.
0.00s - Note: Debugging will proceed. Set PYDEVD_DISABLE_FILE_VALIDATION=1 to disable this validation.
TKOpenGl | Type: Other | ID: 0 | Severity: Medium | Message:
  OpenGl_Window::CreateWindow: window Visual is incomplete: no depth buffer, no stencil buffer
```

jmwright, 2022-04-15:
I'll go ahead and close this issue then.

closed 2022-04-15 by jmwright