CadQuery/CQ-editor #326: Not all lines shown in UI

bullestock, 2022-03-31:
This code
`wire = cq.Workplane("XY") \
         .lineTo(2, 2) \
         .threePointArc((4, 1), (3.5, 0))
`
only shows the arc and not the line resulting from lineTo (see screenshot below).

[image]

This code, however

`wire = cq.Workplane("XY") \
         .lineTo(2, 2) \
         .threePointArc((4, 1), (3.5, 0)) \
         .mirrorX()
`
shows the entire heart shape.

Version: 0.3.0dev, Arch Linux (python-cq-editor-git package).

lorenzncode, 2022-04-01:
CQ-Editor shows the objects of the `Workplane` on the top of the stack.

_As you work in CadQuery, each operation returns a new Workplane object with the result of that operations. Each Workplane object has a list of objects, and a reference to its parent._

 "The Stack" https://cadquery.readthedocs.io/en/latest/index.html

```wire = cq.Workplane("XY").lineTo(2, 2).threePointArc((4, 1), (3.5, 0))```

In this example, there are two calls to `Workplane` methods.  The CQ object inspector shows there are two `Workplane` objects.  The Vector represents the origin of each `Workplane`.

[image]

Expand the tree to show the objects list of each `Workplane`.

`lineTo` adds a single Edge to the objects list.

[image]

`threePointArc` adds a single Edge to the objects list.

[image]

For visualization, `consolidateWires` can be used to combine the two edges (from the modeling context) into a single wire. 

```wire = cq.Workplane("XY").lineTo(2, 2).threePointArc((4, 1), (3.5, 0)).consolidateWires()```

[image]

`mirrorX` automatically calls `consolidateWires` to join edges into a single wire.

bullestock, 2022-04-01:
Guess I should have RTFM.

closed 2022-04-01 by bullestock