CadQuery/CQ-editor #328: Cryptic names in Objects pane

pauljurczak, 2022-04-01:
I'm running CQ-editor Version: 0.2.0 (pre-built package) on Ubuntu 20.04.5. I used to see object names in Object pane corresponding to Python variables, but after system update they disappeared for most files and I see a single numeric value instead, e.g. for this script:
```
import cadquery as cq

boxL = 20.0  
boxW = 20.0  
boxH = 10.0
rout = 2.0
rin = 1.0  
thick = 2.0

oshell = (cq.Workplane("XY").box(boxL, boxW, boxH)
    .edges("|Z").fillet(rout))
ishell = (oshell.faces("<Z").workplane(thick, True)
    .rect((boxW - 2.0*thick), (boxL - 2.0*thick))
    .extrude((boxH - thick), False).edges("|Z").fillet(rin))
box = oshell.cut(ishell)

show_object(box)
cq.exporters.export(box,'bug-box.stl')
```
I see `140124057685296` instead of `oshell` and `ishell`:

[image]

lorenzncode, 2022-04-03:
`show_object` provides an argument for specifying the name (see the README here https://github.com/CadQuery/CQ-editor).

pauljurczak, 2022-04-03:
I used to see `oshell` and `ishell` names there with the same code without specifying name parameter for `show_object`.

pauljurczak, 2022-04-03:
Here is what I mean:

[image]

I see this only in rare cases after random edits.

pauljurczak, 2022-04-03:
The way it behaves now is:
- Python object names in Objects when no `show_object()` is used
- cryptic names when at least one `show_object()` is used

adam-urbanczyk, 2022-04-03:
Please read the readme before opening issues: https://github.com/CadQuery/CQ-editor#naming-an-object

closed 2022-04-03 by adam-urbanczyk

pauljurczak, 2022-04-03:
> Please read the readme before opening issues

I had. The statement there *By default, objects have a randomly generated ID in the object inspector* is not accurate. If `show_object()` is not used, Python variable names are used as object IDs.

[image]