CadQuery/CQ-editor #330: Documentation search issue

pauljurczak, 2022-04-03:
I'm not sure if this can be fixed here, using `close` search query: https://cadquery.readthedocs.io/en/latest/search.html?q=close&check_keywords=yes&area=default
doesn't find `cadquery.Workplane.close()` method:

[image]

adam-urbanczyk, 2022-04-03:
AFAICT the first hit is what you are after.

closed 2022-04-03 by adam-urbanczyk

pauljurczak, 2022-04-03:
< AFAICT the first hit is what you are after.

No. I'm after `cadquery.Workplane.close()`, which is at https://cadquery.readthedocs.io/en/latest/classreference.html?highlight=workplane#cadquery.Workplane.close

[image]

not after `cadquery.Edge.close()` documented at https://cadquery.readthedocs.io/en/latest/classreference.html?highlight=close#cadquery.Edge.close

jmwright, 2022-04-11:
I have the same issue of `Workplane.close()` entry not showing up, but this issue should really be filed in the main cadquery repository.