CadQuery/CQ-editor #331: Enhance show_object to accept string argument

lorenzncode, 2022-04-03:
I typically specify the `show_object` name parameter with the same name as the variable.

```python
show_object(mypart4, name="mypart4")`
```
In this case a descriptive or identifiable variable name has been defined and I'd like to avoid repeating it in the name parameter.
The proposed enhancement is to support a string as the first argument to `show_object`.

```python
show_object("mypart4")
```

The benefit is a less verbose and repetitive syntax.  Editing a script to modify the displayed objects would be (slightly) quicker. 

Of course the existing syntax would continue to work.  Would this change be acceptable?

Example script to show select objects.:

```python
import cadquery as cq

mypart1 = cq.Workplane().box(1, 1, 1)
mypart2 = cq.Workplane(origin=((2, 0, 0))).box(1, 1, 2)
mypart3 = cq.Workplane(origin=((4, 0, 0))).box(1, 1, 3)
mypart4 = cq.Workplane(origin=((6, 0, 0))).box(1, 1, 4)
mypart5 = cq.Workplane(origin=((8, 0, 0))).box(1, 1, 5)
mypart6 = cq.Workplane(origin=((10, 0, 0))).box(1, 1, 6)

show_object("mypart4")
show_object("mypart5")
show_object("mypart6", options={"color": "yellow"})
```

[image]

adam-urbanczyk, 2024-06-16:
Thanks @lorenzncode , in the end solved via #438: `show_object(res)` will result in the name `res` being used in the object tree.

closed 2024-06-16 by adam-urbanczyk

lorenzncode, 2024-06-16:
I gave it a try.  Good feature.  Thanks!