CadQuery/CQ-editor #337: Cannot launch on macOS: invalid code signature in libpython3.8.dylib

fferri, 2022-04-19:
Just downloaded CQ-editor-macOS release from this repository.

When running the CQ-editor.sh script, the following error message appears on macOS:

> “libpython3.8.dylib” cannot be opened because the developer cannot be verified.
> 
> macOS cannot verify that this app is free from malware.
> 
> Safari downloaded this file today at 13:45 from objects.githubusercontent.com.

with options to **Move to trash** or **Cancel**

In the terminal, the following error is printed:

> [7139] Error loading Python lib '/Users/me/Downloads/CQ-editor-MacOS/CQ-editor/libpython3.8.dylib': dlopen: dlopen(/Users/me/Downloads/CQ-editor-MacOS/CQ-editor/libpython3.8.dylib, 10): no suitable image found.  Did find:
> 	/Users/me/Downloads/CQ-editor-MacOS/CQ-editor/libpython3.8.dylib: code signature in (/Users/me/Downloads/CQ-editor-MacOS/CQ-editor/libpython3.8.dylib) not valid for use in process using Library Validation: library load disallowed by system policy

fferri, 2022-04-19:
Solved by:
```
xattr -r -d com.apple.quarantine path/to/CQ-editor-MacOS
```

closed 2022-04-19 by fferri

fferri, 2022-04-19:
No, it is an Intel.
macOS version 10.15.7

subatomicglue, 2023-01-05:
Install instructions in your GitHub for MacOS (and/or) A README.md file in the root of the release `.zip` file for MacOS would be super helpful for this issue:

Should mention the `xattr` here in your GitHub [README.md](https://github.com/CadQuery/CQ-editor/blob/master/README.md) for MacOS versions BigSur and greater (may apply to older ones as well, not sure!):
`On Linux and MacOS you may need to make the script executable with chmod +x CQ-editor.sh `

Since with MacOS's new security schemes every user will be running into this.   They keep track of the .zip file downloaded, even if I rename it, even if I use command line unzip, and mark every file inside it as unsafe... to protect us from potential malware.
I almost deleted and moved on...  (a potential new user was almost lost).

A `dmg` file to run this as an application might be another good idea...

I have MacOS BigSur, Intel...   on a macbookpro 2019.   I was using the latest MacOS `.zip` release.   I have not tried conda or the development package instructions.

MacOS security stuff totally blocks running the script, and each and every .so after that (so it's pretty impossible to manually bypass the security in SystemPreferences UI because >50 clicks I gave up).

Looked around this github and the net for this without finding a great solution (most solutions found involve the unlocking through MacOS ui instead of command line solution).  Until I clicked the "closed" issues here to reveal this one.   This `xattr` help should be easier/faster to find and part of the new user experience.

fferri's fix above worked for me, thanks!