CadQuery/CQ-editor #345: Fails with Ubuntu 22.04 LTS

gumyr, 2022-05-02:
After upgrading Ubuntu to 22.04 LTS, creating a new conda environment, and installing cadquery with `conda install -c conda-forge -c cadquery cadquery=master` and cq-editor with `conda install -c cadquery -c conda-forge cq-editor=master` (nothing else installed), cq-editor comes up with a black window and the following messages:
```
libGL error: MESA-LOADER: failed to open crocus: /usr/lib/dri/crocus_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: crocus
libGL error: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
libGL error: MESA-LOADER: failed to open crocus: /usr/lib/dri/crocus_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: crocus
libGL error: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
```
Investigating I found:
- `libgl1-mesa-dri is already the newest version (22.0.1-1ubuntu2)`
- `libxrender1 is already the newest version (1:0.9.10-1build4)`
- `sudo apt-get install freeglut3-dev` didn't help
- `sudo apt-get install -y libxtst-dev` didn't help
- `sudo apt-get install -y libxft-dev` didn't help

Any assistance would be appreciated.

lorenzncode, 2022-05-03:
It looks like an anaconda issue.  It is similar to #298 which was an issue of the pyinstaller version of CQ-Editor with old libstdc++.  Here it is anaconda with older libstdc++.
[image]

After rm libstdc++ from conda env:
[image]

jmwright, 2022-05-03:
I can confirm that the fix by @lorenzncode does work. The PyInstaller builds run fine on 22.04 because libstdc++ is preemptively removed after the build process completes.

jmwright, 2022-05-03:
I guess I should also note that there are two symlinks in the environment that should be removed. `libstdc++.so.6` and `libstdc++.so`. Once those are removed, CQ-editor should start up fine.

hho, 2022-07-13:
This affects the [current release](https://github.com/CadQuery/CQ-editor/releases/tag/0.2) as well. Looks as in the first screenshot for me. How can I get started with CadQuery now?

jmwright, 2022-07-25:
@hho If you are logged into GitHub, please click on one of the runs [here](https://github.com/jmwright/CQ-editor/actions/workflows/pyinstaller-builds-actions.yml) and download the artifact for your OS. Then please let us know if that development version works for you.

hho, 2022-07-25:
@jmwright Thank you! That works almost flawlessly – only on startup I get a bad color for the highlighting of the last line of the file, but after the first cursor movement everything is perfect.

jmwright, 2022-07-25:
> only on startup I get a bad color for the highlighting of the last line of the file

I've noticed the strange highlight as well with newer builds of CQ-editor. I'm not sure where it is coming from, but I think it started after the last UI update where some button icons seems to have changed.

revast, 2022-10-07:
Ubuntu 22.04:  to use it from zip file download, I had to  `sudo ln -s /usr/lib/x86_64-linux-gnu/dri/ /usr/lib/dri` to avoid
```
libGL error: MESA-LOADER: failed to open vmwgfx: /usr/lib/dri/vmwgfx_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
```

and then remove `libstdc++.so.6` to avoid 

```
libGL error: MESA-LOADER: failed to open vmwgfx: /home/hero/Downloads/CQ-editor-Linux-x86_64/CQ-editor/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/lib/dri/vmwgfx_dri.so) (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: vmwgfx
```

jdegenstein, 2022-10-15:
I tested my static build of my fork of CQ-editor and was able to run on Ubuntu 22.04 without having to remove `libstdc++.so.6`. I think this issue may be resolved because CQ master now depends on OCP 7.6.3, which depends on VTK 9.1.0 which requires GLIBCXX_3.4.30 which is the newest available as far as I can tell (and the same as Ubuntu 22.04).

https://github.com/jdegenstein/jmwright-CQ-Editor/actions

jmwright, 2022-10-17:
Here is a new static build with libstdc++ still bundled if anyone wants to try it.

https://github.com/jmwright/CQ-editor/actions/runs/3259613214

I'll post this to the Arch Linux issue as well.

sambaPython24, 2025-04-25:
@jmwright Since I still have the same problem, would you give us the link again? The current link is broken.