CadQuery/CQ-editor #349: [Suggestion] Add "Render" keyboard shortcut please

qkum, 2022-05-07:
**hello dear fellow Open Source frens.**

please add  a "Render" keyboard shortcut.

i suggest these 4 possibilities based on those commonly used in other python software like jupyter notebook and the fact that the tab is named "Render" :

`["shift" + "r"] or ["ctrl" + "r"] or ["shift" + "enter"] or ["ctrl" + "enter"]`

why? :

when you have gotten used to jupyter notebook shortcuts (used by google colab also etc.) it is a headache/pain/stress factor in your workflow to grab your mouse and click that "Render" button. so big a problem that i actually use 15 min of my time on writing this suggestion in full length. (instead of working on my project).

thanks for your hard work on this awesome project contributors.
thanks for reading my suggestion with a open mind fren 💯

qkum, 2022-05-07:
adding the option to edit this keybinding/hotkey would be even better ofc.

adam-urbanczyk, 2022-05-07:
There is one - `F5`

closed 2022-05-07 by adam-urbanczyk