CadQuery/CQ-editor #350: [Suggestion] App remember settings (changes to the GUI boxes)

qkum, 2022-05-07:
my CQ-editor does not remember the changes to the boxes in the GUI. (grab n drag with mouse to edit the width / height of boxes)

example: 
* "Code" box
* "Console" box
* "Current traceback" box
* "Objects" box
* etc.

i assume it is the same for everybody else.

it may be a bug only for me, but CQ-editor on itself seems to work perfect. (newest version, installed following the github quickstart guide)

my problem is when i try to run CadQuery in Jupyternotebooks in Visual Studio Code Insider Notebooks (maybe the problem is the Insider version that I use - it suddenly hit me) 

"Visual Studio Code Insider" is just the cutting edge version the normal "Visual Studio Code" for those who do not know.

- i told you the last part in case you read my other posts talking about the error that im currently battling with.

thanks for reading my suggestion with a open mind fren 💯🐸

adam-urbanczyk, 2022-05-07:
It is implemented, more details on your env?

SeanDS, 2022-05-15:
I was going to open an issue but this might already cover it. One thing the "objects" box forgets is each time between runs the displayed objects get reset to "on", regardless of their previous state:

[image]

It'd be nice if the checkbox states were preserved between runs, so long as the displayed objects haven't been modified in the code.

adam-urbanczyk, 2022-05-17:
AFAIR it is already there:
[image]

SeanDS, 2022-05-17:
> AFAIR it is already there: [image]

Aha yes, that's it! Thanks.

closed 2025-03-06 by jmwright