CadQuery/CQ-editor #351: Possible Dangerous Interaction - ctrl+s deleting work instead of saving

1knueller, 2022-05-14:
Hi!
Im using CQ Editor with an external editor. I have autoreload on. and I have pycharm side by side with the cq editor open. so when i make changes, i hit save in the pycharm editor and i see the results. i have closed the codewindow of cq editor.

I found a frightening problem. When I did some nice work, being happy with myself I saved the file in the IDE and then closed the IDE. Then i clicked into cq editor (into the render) and - because im used to do that - i hit ctrl+s and close. 

Then I find my work file had 0kb. Turns out it will be overwritten by an empty file when clicking into the render and clicking ctrl+s.
I also found it to overwrite it with an empty file when clicking in the code editor and hit ctrl+s. After the previous rocedere.

Very lucky I worked with pycharm that day because it stores a local file history wich made it possible to restore nearly everything except for the last few lines i wrote.

Im using cadquery from a anaconda install:
installed like so:
conda install -c cadquery -c conda-forge cq-editor=master

cadquery                  master                   py3.10    cadquery
cq-editor                 master                   py3.10    cadquery

lorenzncode, 2022-05-14:
`Ctrl + S` will save the contents of the Editor to a file.  The result is the same whether the focus is in the Editor or viewer or if the Editor widget is hidden or visible.

I cannot reproduce the problem where the Editor is non-empty and `Ctrl + S` writes an empty file.

> I also found it to overwrite it with an empty file when clicking in the code editor and hit ctrl+s.

What was the contents of the Editor when the empty file was saved?

1knueller, 2022-05-14:
My code. I looked at it. It was there. I hit CTRL+S with the editor focused and i see the file in the explorer go to 0kb and when click into pycharm it reloads from disk and the pycharm editor will be emptied aswell.

It didnt matter what I wrote into the cqeditor code window.

I will try to reproduce and record it again later.

1knueller, 2022-05-14:
Hello!

I could reproduce the issue.
https://cadquery.readthedocs.io/en/latest/examples.html#braille-example

steps to reproduce: 
Create new python project in pycharm with the example main file.
remove main file contents
place code from braille example
import cq
save
focus windows terminal
activate cadquery conda environment
open cq editor
* autoreload is already activated
open the file we previously generated through menu
build the object
add some random comment through pycharm editor to verify cqeditor is reloading the file
focus cqeditor
ctrl+s
see the file on disk has 0kb

I also tried renaming the file to not be main.
From this point on everything is bugged. The autoreload also does not do anything anymore. So if i write into pycharm editor and save cqeditor doesnt do anything. it just shows the previous code.

willhansen, 2022-06-05:
Might be a good idea to have a "read only" option next to the "autoreload" option?

It would prevent mishaps for people exclusively using external editors, but still allow debugging in CQ-editor.

This is the behavior the Arduino editor uses with the "external editor" option enabled.

voneiden, 2022-07-04:
I'm willing to bet my hat that this is an encoding issue and related to  #276 . The braille example has utf8 encoded braille characters which require Unicode compatible encoding when saving. 

When loading, the underlying spyder CodeEditor handles encoding detection and decodes the data correctly. 

When saving, CQ-Editor opens the file for writing in text mode. This clears the file and it uses system default encoding. It then attempts to write the editor data into the file. This leads to an automatic encoding step using the system default encoding and it will fail if the system default encoding does not know how to encode those special characters.

https://github.com/CadQuery/CQ-editor/blob/master/cq_editor/widgets/editor.py#L169

Appropriate behaviour would be to check file encoding before saving (if it exists) and use the same encoding.

voneiden, 2022-07-04:
I've reproduced this by forcing a non-utf8 encoding

```
>>> sys.getfilesystemencoding()
'iso8859-1'
```

File is cleared and CQ-Editor log viewer shows

```
[2022-07-04 14:38:52.523134] ERROR: CQ-Editor: Uncaught exception occurred
Traceback (most recent call last):
  File "../miniforge3/lib/python3.10/site-packages/cq_editor/widgets/editor.py", line 170, in save
    f.write(self.toPlainText())
UnicodeEncodeError: 'latin-1' codec can't encode character '\u2820' in position 187: ordinal not in range(256)
```

Relevant reading:

https://peps.python.org/pep-0597/#using-the-default-encoding-is-a-common-mistake
https://peps.python.org/pep-0686/ UTF-8 will become default instead of system encoding in Python 3.15

1knueller, 2022-07-09:
Hi!

Testing your commit I could not reproduce the issue anymore. 

I tried on the master with your most recent variant of the file (commit 826a7b14a3022e62a01dedabf1c27a4fcc8ac291 with the encoded outside the with file)

Well done! Thanks!

closed 2022-07-12 by adam-urbanczyk