CadQuery/CQ-editor #353: default mouse rotating left/right rotates view dependent, should rotate around world y-Axis

KadaB, 2022-06-27:
Hey,

I think the mouse orientation in the 3d view is 90% right, except way the mouse rotates around the y-Axis (pulling the mouse left/right). This isn't really intuitive and actually quite confusing. The camera rotates around the camera up vector, but should rotate around the world up (or global y-Axis) vector.

Either way, there should be an option to let the viewer decide if he/she wants to rotate around the view dependent camera up vector or the world coordinate system up vector (or Y-Axis).

KadaB, 2022-06-28:
In [occt_widget.py Line 91](https://github.com/CadQuery/CQ-editor/blob/4b461fe195d0a4e99b9a6c43b7e1fe0cb4c5e77d/cq_editor/widgets/occt_widget.py#L91) I've found the rotation method called from the V3d_Viewer. [V3d_Viewer](https://dev.opencascade.org/doc/occt-6.9.1/refman/html/class_v3d___view.html) has multiple [rotation methods](https://dev.opencascade.org/doc/occt-6.9.1/refman/html/class_v3d___view.html#a7a6cae8fdac0a242be51300920605f8e)
```
        if event.buttons() == Qt.LeftButton:
            self.view.Rotation(x,y)
```

Maybe one of the four with the right parameters can rotate the view around the world up axis. I'll have to check, but it seems that it isn't too hard to change :-)

bendavis78, 2023-09-24:
Hi, did you find a solution to this? I find the orbit style very clunky. I much prefer OpenSCAD's mouse controls.

bendavis78, 2023-09-24:
@KadaB, did you find a solution to this? I find the orbit style very clunky. I much prefer OpenSCAD's mouse controls.

zwn, 2024-02-16:
There is a PR #220 adding a different style of turning/orbiting. Is that what you are after? I'd prefer to have the turn table method as well (see the videos in the PR).

zwn, 2024-02-16:
It is also the way the github builtin viewer for stl files use. See for example https://github.com/openvmp/openvmp-models/blob/main/generated_files/parts/arduino/mega2560.stl and try to rotate the view there.

xahon, 2025-05-06:
Agreed with OP, the view camera should orbit around the selected object with Z-axis lock instead of unconstrained rotation (which is not usable IMO). I also think fly-camera would be quite useful to see object from inside and have a convenient WASD(QE) navigation like a spectator mode in many games

jmwright, 2025-05-06:
I prefer the way the rotation currently works, but I can understand wanting other rotation schemes. If someone creates a PR for this feature, please add a setting for it to the preferences dialog and keep the current rotation behavior as the default.

xahon, 2025-05-06:
> I prefer the way the rotation currently works, but I can understand wanting other rotation schemes. If someone creates a PR for this feature, please add a setting for it to the preferences dialog and keep the current rotation behavior as the default.

I tried to apply this PR with some changes but completely lost in how `restorePreferences` works. It caches values in windows registry and then wipes out after a lot of signals call `restorePreferences` over and over again. Tried to change `QSettings` to use ini format and clear it out but those signals secretly wipe out default values in preferences map and I couldn't figure out what causes it. I gave up after then

jmwright, 2025-07-07:
The turntable orbit method had been merged. 

https://github.com/CadQuery/CQ-editor/pull/525