CadQuery/CQ-editor #358: multimethod.DispatchError when option Reload CQ enabled

lorenzncode, 2022-07-16:
To reproduce:
- Install current master
- Open CQ-editor with code
   ```
   wire0 = cq.Workplane().lineTo(5, 5).lineTo(10, 0).close().val()
    
   # solid0 = cq.Solid.extrudeLinear(cq.Face.makeFromWires(wire0), (0, 0, 1))
   solid1 = cq.Solid.extrudeLinear(cq.Face.makeFromWires(wire0), cq.Vector(0, 0, 1))
    
   # r0 = cq.Workplane(solid0)
   r1 = cq.Workplane(solid1).translate((10, 0, 0))
   ```
- Edit->Preferences->Debugger
   Select "Reload CQ"
- Render two times
  The traceback in the UI shows:
   ```TypeError: Solid.extrudeLinear() missing 1 required positional argument: 'vecNormal'```

   The full traceback includes a DispatchError.

   ```multimethod.DispatchError: ('extrudeLinear: 0 methods found', (<class 'type'>, <class 'cadquery.occ_impl.shapes.Face'>, <class 'cadquery.occ_impl.geom.Vector'>), [])```

adam-urbanczyk, 2022-07-17:
Hm, I cannot reproduce. Could you provide all details about your env?

lorenzncode, 2022-07-17:
Here are the env details.

<details>
<summary>env</summary>

<pre>
# packages in environment at /home/lorenzn/tools/miniforge_apr26/envs/cqjul17b:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
alabaster                 0.7.12                     py_0    conda-forge
alsa-lib                  1.2.3.2              h166bdaf_0    conda-forge
ampl-mp                   3.1.0             h2cc385e_1006    conda-forge
arrow                     1.2.2              pyhd8ed1ab_0    conda-forge
astroid                   2.11.6          py310hff52083_0    conda-forge
atomicwrites              1.4.1              pyhd8ed1ab_0    conda-forge
attrs                     21.4.0             pyhd8ed1ab_0    conda-forge
autopep8                  1.6.0              pyhd8ed1ab_1    conda-forge
babel                     2.10.3             pyhd8ed1ab_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
beautifulsoup4            4.11.1             pyha770c72_0    conda-forge
binaryornot               0.4.4                      py_1    conda-forge
black                     22.6.0             pyhd8ed1ab_0    conda-forge
bleach                    5.0.1              pyhd8ed1ab_0    conda-forge
brotlipy                  0.7.0           py310h5764c6d_1004    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.18.1               h7f98852_0    conda-forge
ca-certificates           2022.6.15            ha878542_0    conda-forge
cadquery                  master                   py3.10    cadquery
casadi                    3.5.5           py310hb2c22f7_11    conda-forge
certifi                   2022.6.15       py310hff52083_0    conda-forge
cffi                      1.15.1          py310h255011f_0    conda-forge
chardet                   5.0.0           py310hff52083_0    conda-forge
charset-normalizer        2.1.0              pyhd8ed1ab_0    conda-forge
click                     8.1.3           py310hff52083_0    conda-forge
cloudpickle               2.1.0              pyhd8ed1ab_0    conda-forge
colorama                  0.4.5              pyhd8ed1ab_0    conda-forge
cookiecutter              2.1.1              pyh6c4a22f_0    conda-forge
cq-editor                 master                   py3.10    cadquery
cryptography              37.0.4          py310h597c629_0    conda-forge
curl                      7.83.1               h7bff187_0    conda-forge
dataclasses               0.8                pyhc8e2a94_3    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
debugpy                   1.6.0           py310hd8f1fbe_0    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
diff-match-patch          20200713           pyh9f0ad1d_0    conda-forge
dill                      0.3.5.1            pyhd8ed1ab_0    conda-forge
docutils                  0.18.1          py310hff52083_1    conda-forge
double-conversion         3.1.7                h9c3ff4c_0    conda-forge
eigen                     3.4.0                h4bd325d_0    conda-forge
entrypoints               0.4                pyhd8ed1ab_0    conda-forge
expat                     2.4.8                h27087fc_0    conda-forge
ezdxf                     0.17.2          py310hbf28c38_1    conda-forge
ffmpeg                    4.3.2                h37c90e5_3    conda-forge
flake8                    4.0.1              pyhd8ed1ab_2    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.0               h8e229c2_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freeimage                 3.18.0               hf18588b_9    conda-forge
freetype                  2.10.4               h0708190_1    conda-forge
gettext                   0.19.8.1          h73d1719_1008    conda-forge
gl2ps                     1.4.2                h0708190_0    conda-forge
[... 226 lines cut ...]
x264                      1!161.3030           h7f98852_1    conda-forge
xorg-kbproto              1.0.7             h7f98852_1002    conda-forge
xorg-libice               1.0.10               h7f98852_0    conda-forge
xorg-libsm                1.2.3             hd9c2040_1000    conda-forge
xorg-libx11               1.7.2                h7f98852_0    conda-forge
xorg-libxau               1.0.9                h7f98852_0    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xorg-libxext              1.3.4                h7f98852_1    conda-forge
xorg-libxt                1.2.1                h7f98852_2    conda-forge
xorg-xextproto            7.3.0             h7f98852_1002    conda-forge
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
[... cut ...]
yapf                      0.32.0             pyhd8ed1ab_0    conda-forge
zeromq                    4.3.4                h9c3ff4c_1    conda-forge
zipp                      3.8.0              pyhd8ed1ab_0    conda-forge
zlib                      1.2.12               h166bdaf_2    conda-forge
zstd                      1.5.2                h8a70e8d_2    conda-forge
</pre>

<pre>
Linux fedora 5.18.9-100.fc35.x86_64 #1 SMP PREEMPT_DYNAMIC Sat Jul 2 15:57:56 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
</pre>

</details>

lorenzncode, 2022-07-17:
I tried again with a clean install on Windows and could reproduce in that env too.

<details>
<summary>Windows env</summary>

<pre>
# packages in environment at C:\projects\cq-editor-358\mambaforge\envs\cqjul17:
#
# Name                    Version                   Build  Channel
alabaster                 0.7.12                     py_0    conda-forge
arrow                     1.2.2              pyhd8ed1ab_0    conda-forge
astroid                   2.11.6          py310h5588dad_0    conda-forge
atomicwrites              1.4.1              pyhd8ed1ab_0    conda-forge
attrs                     21.4.0             pyhd8ed1ab_0    conda-forge
autopep8                  1.6.0              pyhd8ed1ab_1    conda-forge
babel                     2.10.3             pyhd8ed1ab_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
bcrypt                    3.2.2           py310he2412df_0    conda-forge
beautifulsoup4            4.11.1             pyha770c72_0    conda-forge
binaryornot               0.4.4                      py_1    conda-forge
black                     22.6.0             pyhd8ed1ab_0    conda-forge
bleach                    5.0.1              pyhd8ed1ab_0    conda-forge
brotlipy                  0.7.0           py310he2412df_1004    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2022.6.15            h5b45459_0    conda-forge
cadquery                  master                   py3.10    cadquery
casadi                    3.5.5           py310he93d0b7_11    conda-forge
certifi                   2022.6.15       py310h5588dad_0    conda-forge
cffi                      1.15.1          py310hcbf9ad4_0    conda-forge
chardet                   5.0.0           py310h5588dad_0    conda-forge
charset-normalizer        2.1.0              pyhd8ed1ab_0    conda-forge
click                     8.1.3           py310h5588dad_0    conda-forge
cloudpickle               2.1.0              pyhd8ed1ab_0    conda-forge
colorama                  0.4.5              pyhd8ed1ab_0    conda-forge
cookiecutter              2.1.1              pyh6c4a22f_0    conda-forge
cq-editor                 master                   py3.10    cadquery
cryptography              37.0.4          py310ha857299_0    conda-forge
curl                      7.83.1               h789b8ee_0    conda-forge
dataclasses               0.8                pyhc8e2a94_3    conda-forge
debugpy                   1.6.0           py310h8a704f9_0    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
diff-match-patch          20200713           pyh9f0ad1d_0    conda-forge
dill                      0.3.5.1            pyhd8ed1ab_0    conda-forge
docutils                  0.18.1          py310h5588dad_1    conda-forge
double-conversion         3.1.7                h0e60522_0    conda-forge
eigen                     3.4.0                h2d74725_0    conda-forge
entrypoints               0.4                pyhd8ed1ab_0    conda-forge
expat                     2.4.8                h39d44d4_0    conda-forge
ezdxf                     0.17.2          py310h476a331_1    conda-forge
ffmpeg                    4.3.1                ha925a31_0    conda-forge
flake8                    4.0.1              pyhd8ed1ab_2    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.0               hce3cb01_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freeimage                 3.18.0               h6676e37_9    conda-forge
freetype                  2.10.4               h546665d_1    conda-forge
gl2ps                     1.4.2                h0597ee9_0    conda-forge
glew                      2.1.0                h39d44d4_2    conda-forge
hdf4                      4.2.15               h0e5069d_3    conda-forge
hdf5                      1.10.6          nompi_h5268f04_1114    conda-forge
icu                       69.1                 h0e60522_0    conda-forge
idna                      3.3                pyhd8ed1ab_0    conda-forge
imagesize                 1.4.1              pyhd8ed1ab_0    conda-forge
[... 203 lines cut ...]
                          C:\Users\lorenzn\.condarc
          conda version : 4.12.0
    conda-build version : not installed
         python version : 3.9.13.final.0
       virtual packages : __win=0=0
                          __archspec=1=x86_64
       base environment : C:\projects\cq-editor-358\mambaforge  (writable)
      conda av data dir : C:\projects\cq-editor-358\mambaforge\etc\conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/win-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : C:\projects\cq-editor-358\condacache
       envs directories : C:\projects\cq-editor-358\mambaforge\envs
                          C:\Users\lorenzn\.conda\envs
                          C:\Users\lorenzn\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/4.12.0 requests/2.27.1 CPython/3.9.13 Windows/10 Windows/10.0.19044
          administrator : False
             netrc file : None
           offline mode : False
</pre>

</details>

[image]

closed 2022-07-20 by adam-urbanczyk

lorenzncode, 2023-01-27:
FYI, I tested with Python 3.11 and can reproduce this issue again. 

<details>
<summary>env</summary>
<pre>
cookiecutter              2.1.1              pyh6c4a22f_0    conda-forge
cq-editor                 0.3.0.dev0                dev_0    <develop>
cryptography              39.0.0          py311h9b4c7bb_0    conda-forge
curl                      7.87.0               hdc1c0ab_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
debugpy                   1.6.6           py311hcafe171_0    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
diff-match-patch          20200713           pyh9f0ad1d_0    conda-forge
dill                      0.3.6              pyhd8ed1ab_1    conda-forge
docstring-to-markdown     0.11               pyhd8ed1ab_0    conda-forge
docutils                  0.19            py311h38be061_1    conda-forge
double-conversion         3.2.0                h27087fc_1    conda-forge
eigen                     3.4.0                h4bd325d_0    conda-forge
entrypoints               0.4                pyhd8ed1ab_0    conda-forge
exceptiongroup            1.1.0              pyhd8ed1ab_0    conda-forge
executing                 1.2.0              pyhd8ed1ab_0    conda-forge
expat                     2.5.0                h27087fc_0    conda-forge
ezdxf                     1.0.0           py311ha3edf6b_0    conda-forge
ffmpeg                    5.1.2           gpl_h8dda1f0_106    conda-forge
fftw                      3.3.10          nompi_hf0379b8_106    conda-forge
flake8                    6.0.0              pyhd8ed1ab_0    conda-forge
flit-core                 3.8.0              pyhd8ed1ab_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.1               hc2a2eb6_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.38.0          py311hd4cff14_1    conda-forge
freeimage                 3.18.0              h530e30e_10    conda-forge
freetype                  2.12.1               hca18f0e_1    conda-forge
frozenlist                1.3.3           py311hd4cff14_0    conda-forge
gettext                   0.21.1               h27087fc_0    conda-forge
giflib                    5.2.1                h36c2ea0_2    conda-forge
gl2ps                     1.4.2                h0708190_0    conda-forge
glew                      2.1.0                h9c3ff4c_2    conda-forge
glib                      2.74.1               h6239696_1    conda-forge
glib-tools                2.74.1               h6239696_1    conda-forge
gmp                       6.2.1                h58526e2_0    conda-forge
gnutls                    3.7.8                hf3e180e_0    conda-forge
graphite2                 1.3.13            h58526e2_1001    conda-forge
gst-plugins-base          1.21.3               h4243ec0_1    conda-forge
gstreamer                 1.21.3               h25f0c4b_1    conda-forge
gstreamer-orc             0.4.33               h166bdaf_0    conda-forge
harfbuzz                  6.0.0                h8e241bc_0    conda-forge
hdf4                      4.2.15               h9772cbc_5    conda-forge
hdf5                      1.12.2          nompi_h4df4325_101    conda-forge
hunter                    3.5.1           py311ha362b79_1    conda-forge
icu                       70.1                 h27087fc_0    conda-forge
idna                      3.4                pyhd8ed1ab_0    conda-forge
imagesize                 1.4.1              pyhd8ed1ab_0    conda-forge
imath                     3.1.6                h6239696_1    conda-forge
importlib-metadata        6.0.0              pyha770c72_0    conda-forge
importlib_metadata        6.0.0                hd8ed1ab_0    conda-forge
importlib_resources       5.10.2             pyhd8ed1ab_0    conda-forge
inflection                0.5.1              pyh9f0ad1d_0    conda-forge
iniconfig                 2.0.0              pyhd8ed1ab_0    conda-forge
intervaltree              3.0.2                      py_0    conda-forge
ipopt                     3.14.10              hf60ba86_0    conda-forge
ipykernel                 6.20.2             pyh210e3f2_0    conda-forge
ipython                   8.8.0              pyh41d4057_0    conda-forge
ipython_genutils          0.2.0                      py_1    conda-forge
[... 288 lines cut ...]
xorg-libx11               1.7.2                h7f98852_0    conda-forge
xorg-libxau               1.0.9                h7f98852_0    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xorg-libxext              1.3.4                h7f98852_1    conda-forge
xorg-libxfixes            5.0.3             h7f98852_1004    conda-forge
xorg-libxrender           0.9.10            h7f98852_1003    conda-forge
xorg-libxt                1.2.1                h7f98852_2    conda-forge
xorg-renderproto          0.11.1            h7f98852_1002    conda-forge
[... cut ...]
yarl                      1.8.2           py311hd4cff14_0    conda-forge
zeromq                    4.3.4                h9c3ff4c_1    conda-forge
zipp                      3.11.0             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               h166bdaf_4    conda-forge
zstd                      1.5.2                h3eb15da_6    conda-forge

</pre>
<pre>

Linux fedora 6.1.6-100.fc36.x86_64 #1 SMP PREEMPT_DYNAMIC Sat Jan 14 17:00:40 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
</pre>
</details>

PR #359 (2022-07-20): CQ reload fix
Closes #358