CadQuery/CQ-editor #363: openGl error and crash Manjaro

ecjohnsoniv, 2022-08-20:
I have been trying to use CadQuery with the CQ-editor, I have installed both with conda and via the pip install and binary method detailed in the Readme. Every time I get everything installed I have the same problem, it gives me the following error:

'
TKOpenGl | Type: Other | ID: 0 | Severity: Medium | Message:
  OpenGl_Window::CreateWindow: window Visual is incomplete: no stencil buffer
' 
The editor will initially load load basic objects like a box, but if you add a small degree of complexity it will freeze and crash. I have updated my drivers according to my drivers and run run benchmaks without problem. I have also googled to no avail and scowered the arch and openGl wikis.

CPU: AMD Ryzen 5 2600 (12) @ 3.400GHz
GPU: NVIDIA GeForce GTX 1070

jmwright, 2022-08-20:
Which operating system?

jmwright, 2022-08-20:
NVM, it's in the title.

lorenzncode, 2022-08-20:
I cannot reproduce a crash on Manjaro.  Could you upload the problematic script?

adam-urbanczyk, 2022-08-21:
The warning is likely not related to the issue. I get it on my system too.

ecjohnsoniv, 2022-08-21:
This script crashed the program

`
import cadquery as cq

def winger(size):
    x = size * 5
    y = size * 1
    z = size * 3
    result = cq.Workplane().box(x, y, z, centered=False)
    #tube = cq.Workplane().cylinder(1, 5)
    #result = result.union(tube)
    return result

final = cq.Workplane()

many = 3
for i in range(many):
    new = winger(5).rotate([0,0,0], [0,0,1], i*(360/many))
    final = result.union(new)

tube = cq.Workplane().cylinder(1, 5)

#result = winger(5)
show_object(final)
show_object(tube)

`

lorenzncode, 2022-08-22:
I took a guess at the indentation ([To format code or text into its own distinct block, use triple backticks](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#quoting-code)) and removed the comments.

I also changed this line from:
`final = result.union(new)`
to
`final = final.union(new)`
in order to resolve _NameError: name 'result' is not defined_

I could not reproduce the crash with the following:

```python

import cadquery as cq

def winger(size):
    x = size * 5
    y = size * 1
    z = size * 3
    result = cq.Workplane().box(x, y, z, centered=False)
    return result

final = cq.Workplane()

many = 3
for i in range(many):
    new = winger(5).rotate([0,0,0], [0,0,1], i*(360/many))
    final = final.union(new)

tube = cq.Workplane().cylinder(1, 5)

show_object(final)
show_object(tube)
```

If rendering is slow for some models, and consumes large memory 
[(see  issue 286 not a crash)](https://github.com/CadQuery/CQ-editor/issues/286#issuecomment-906755575), you can try increasing the Deviation value in CQ-Editor Preferences.  See 3DViewer->Deviation.  The default is 1e-05.  Try say 1e-3.

ecjohnsoniv, 2022-08-25:
Ok, will do. Thanks for the help.

closed 2025-02-11 by jmwright