CadQuery/CQ-editor #366: mkColor error on launching CQ-editor

lorenzncode, 2022-10-01:
I created a new install today and find CQ-editor fails to launch.

<details>
<summary>env</summary>

<pre>
# packages in environment at /home/lorenzn/tools/miniforge_sep21/envs/cqdev:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
alabaster                 0.7.12                     py_0    conda-forge
alsa-lib                  1.2.7.2              h166bdaf_0    conda-forge
ampl-mp                   3.1.0             h2cc385e_1006    conda-forge
anyio                     3.6.1              pyhd8ed1ab_1    conda-forge
aom                       3.4.0                h27087fc_1    conda-forge
argon2-cffi               21.3.0             pyhd8ed1ab_0    conda-forge
argon2-cffi-bindings      21.2.0          py310h5764c6d_2    conda-forge
arrow                     1.2.3              pyhd8ed1ab_0    conda-forge
astroid                   2.12.10         py310hff52083_0    conda-forge
atomicwrites              1.4.1              pyhd8ed1ab_0    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
attrs                     22.1.0             pyh71513ae_1    conda-forge
autopep8                  1.6.0              pyhd8ed1ab_1    conda-forge
babel                     2.10.3             pyhd8ed1ab_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
beautifulsoup4            4.11.1             pyha770c72_0    conda-forge
binaryornot               0.4.4                      py_1    conda-forge
black                     22.8.0          py310hff52083_0    conda-forge
bleach                    5.0.1              pyhd8ed1ab_0    conda-forge
brotli                    1.0.9                h166bdaf_7    conda-forge
brotli-bin                1.0.9                h166bdaf_7    conda-forge
brotlipy                  0.7.0           py310h5764c6d_1004    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.18.1               h7f98852_0    conda-forge
ca-certificates           2022.9.24            ha878542_0    conda-forge
cadquery                  0.0.0                     dev_0    <develop>
casadi                    3.5.5           py310hc77112e_12    conda-forge
certifi                   2022.9.24          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1          py310h255011f_0    conda-forge
chardet                   5.0.0           py310hff52083_0    conda-forge
charset-normalizer        2.1.1              pyhd8ed1ab_0    conda-forge
click                     8.1.3           py310hff52083_0    conda-forge
cloudpickle               2.2.0              pyhd8ed1ab_0    conda-forge
colorama                  0.4.5              pyhd8ed1ab_0    conda-forge
contourpy                 1.0.5           py310hbf28c38_0    conda-forge
cookiecutter              2.1.1              pyh6c4a22f_0    conda-forge
cq-editor                 0.3.0.dev0                dev_0    <develop>
cryptography              37.0.4          py310h597c629_0    conda-forge
curl                      7.83.1               h7bff187_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
debugpy                   1.6.3           py310hd8f1fbe_0    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
diff-match-patch          20200713           pyh9f0ad1d_0    conda-forge
dill                      0.3.5.1            pyhd8ed1ab_0    conda-forge
docutils                  0.19            py310hff52083_0    conda-forge
double-conversion         3.2.0                h27087fc_1    conda-forge
eigen                     3.4.0                h4bd325d_0    conda-forge
entrypoints               0.4                pyhd8ed1ab_0    conda-forge
expat                     2.4.9                h27087fc_0    conda-forge
ezdxf                     0.18.1          py310hbf28c38_0    conda-forge
ffmpeg                    4.4.2           gpl_hfe78399_107    conda-forge
fftw                      3.3.10          nompi_hf0379b8_105    conda-forge
flake8                    4.0.1              pyhd8ed1ab_2    conda-forge
flit-core                 3.7.1              pyhd8ed1ab_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
[... 393 lines cut ...]
@@ -32,7 +32,7 @@ class ObjectTreeItem(QTreeWidgetItem):
                  shape_display=None,
                  sig=None,
                  alpha=0.,
-                 color='f4a824',
+                 color='#f4a824',
                  **kwargs):
 
         super(ObjectTreeItem,self).__init__([name],**kwargs)
diff --git a/cq_editor/widgets/viewer.py b/cq_editor/widgets/viewer.py
index f788fb1..2cf2d64 100644
--- a/cq_editor/widgets/viewer.py
+++ b/cq_editor/widgets/viewer.py
@@ -38,7 +38,7 @@ class OCCViewer(QWidget,ComponentMixin):
         {'name': 'Use gradient', 'type': 'bool', 'value': False},
         {'name': 'Background color', 'type': 'color', 'value': (95,95,95)},
         {'name': 'Background color (aux)', 'type': 'color', 'value': (30,30,30)},
-        {'name': 'Default object color', 'type': 'color', 'value': "FF0"},
+        {'name': 'Default object color', 'type': 'color', 'value': "#FF0"},
         {'name': 'Deviation', 'type': 'float', 'value': 1e-5, 'dec': True, 'step': 1},
         {'name': 'Angular deviation', 'type': 'float', 'value': 0.1, 'dec': True, 'step': 1},
         {'name': 'Projection Type', 'type': 'list', 'value': 'Orthographic',
```

See also https://pyqtgraph.readthedocs.io/en/latest/functions.html#pyqtgraph.mkColor

lorenzncode, 2022-10-01:
https://github.com/pyqtgraph/pyqtgraph/commit/895be36cd7bfc4dbc5d22f288b539085cf789a9f#diff-c5aa5c9712339df6df9e9ee044ca1dfb4a5b6d3e305fb996a36f15cc6d00fb25

```py 
 warnings.warn(
                    "Parsing of hex strings that do not start with '#' is"
                    "deprecated and support will be removed in 0.13",
                    DeprecationWarning, stacklevel=2
                )
```

kleinee, 2022-10-03:
@lorenzncode: I ran into the same issue on a fresh install. Made your suggested changes in the code manually and confirm they work. Had the same suspicion that "#" were missing. Thanks for hunting this down!

Eckhard

roseengineering, 2022-10-03:
Using pyqtgraph version 0.13.1 from conda-forge breaks (for colors without the # prefix) but version 0.12.3 from pkgs/main works.

PR #367 (2022-10-07): Prefix hex color strings with '#' (#366)
Fixes #366.
closed 2022-10-07 by adam-urbanczyk