CadQuery/CQ-editor #370: 3D viewer image does not fill frame after resizing window

billtubbs, 2022-10-10:
## To Reproduce

I am launching CQ-editor from the console
```console
CQ-editor
```

When I resize the window (or change the proportions of the internal frames within the window) the rendered 3D image does not adjust to the new frame size.

[image]

If I quit QT-editor, reopen and then render it fills the window.

Not sure if this is related to #369.

## Environment

OS X 10.15.7
```
conda install -c cadquery -c conda-forge cq-edimaster
```

adam-urbanczyk, 2022-10-10:
I don't have a mac, so it'll be difficult to solve. What happens if you try to interact with the canvas (drag or click on it)? Is anything weird reported to stdout or the logging window?

billtubbs, 2022-10-10:
The canvas itself works fine.  I can drag and rotate the view no problems.  I don't see any messages in the current traceback or log viewer windows, or in the console.

billtubbs, 2022-10-10:
Interestingly, if I shrink the window to a smaller size, the view descends down off the bottom of the frame leaving a black space above it.

[image]