CadQuery/CQ-editor #373: CQ-Editor Color/Material Mismatch

jdegenstein, 2022-10-20:
In reference to https://github.com/CadQuery/CQ-editor/pull/371, it seems that the new jade material is not used when a custom color/alpha is used. I would like jade to become the default material since the surface reflectivity of the brass material is pretty high.

```py
res1 = cq.Workplane('XY').box(5, 7, 5) 
res2 = cq.Workplane('XY').box(8, 5, 4)
show_object(res1)                     #default material
show_object(res2,options={"alpha":0}) #more "brown" and higher surface reflectivity
```
[image]

I think the "brown" object is a result of show_object not using the same material whenever a non-empty dict is passed to show_object(options={}).

My guess is that the default material should be used regardless of a custom transparency/color. Also, the surface reflectivity of `res2` above is much higher than `res1`. You can see this when looking at e.g. top view.

adam-urbanczyk, 2022-10-20:
Hm, seems some magic is going on within OCCT. I'll set the material explicitly for every AIS object.

closed 2022-10-27 by adam-urbanczyk

jdegenstein, 2022-10-27:
Thanks! Very excited to try this -- and I am sure others in the community are as well.

PR #375 (2022-10-27): Transparency/material fix
Fixes #373