CadQuery/CQ-editor #376: [question] Where does cq-editor store path name and file name of current source file ??

Steve-Petrie, 2022-10-27:
Using CQ-editor-0.2 on Debian GNU/Linux 9 (stretch) Linux 4.9.0-6-amd64.

I installed this pre-built cq-editor from file CQ-editor-Linux-x86_64.zip on 15 FEB 2021.

I start cq-editor with shell command:
/home/steve/apps/CQ-editor-0.2/CQ-editor/CQ-editor

When cq-editor starts, it **reloads the python source file that was open when cq-editor was last closed**, for example:
/home/steve/Bikeway/cq2/bwy_BikewayLoXEndSection.py

**Question: In what file (path name & file name) does cq-editor store the path name and file name of the current source file, so the source file can be reloaded by cq-editor the next time it starts ??**

jmwright, 2022-10-27:
@adam-urbanczyk will know for sure, but I believe CQ-editor uses QSettings for that sort of thing.

https://doc.qt.io/qtforpython-5/PySide2/QtCore/QSettings.html

Here are all the locations the settings file can be stored.

https://doc.qt.io/qtforpython-5/PySide2/QtCore/QSettings.html#locations-where-application-settings-are-stored

Steve-Petrie, 2022-10-27:
**Jeremy,**

Thanks for your prompt and helpful response, that guided me to the solution.

I found the QSettings file:

**/home/steve/.config/CadQuery/CQ GUI.conf**

that contains the text:
...
**[Code%20Editor]
state=/home/steve/Bikeway/cq2/bwy_BikewayLoXEndSection.py**
...

closed 2022-10-27 by Steve-Petrie