CadQuery/CQ-editor #381: The sample code, opened at first start is ASCII, should be UTF-8

LightTemplar, 2022-11-17:
1. I installed cq-editor and cadquery
2. started cq-editor for the first time
3. it showed me some sample code. 
4. I changed it, added some symbols in my native language as comment
5. and tried to save.
6. CQ-Editor threw an Error:
```
[2022-11-17 03:24:43.249238] ERROR: CQ-Editor: Uncaught exception occurred
Traceback (most recent call last):
  File "/home/alexander/Applications/anaconda3/envs/cad/lib/python3.10/site-packages/cq_editor/widgets/editor.py", line 180, in save
    encoded = self.toPlainText().encode(encoding)
UnicodeEncodeError: 'ascii' codec can't encode characters in position 144-147: ordinal not in range(128)
```
If I do the same in a new file, it works without such problem.

I suppose, it's because this first sample code was in ASCII, but should be in UTF-8 encoding, according to https://github.com/CadQuery/CQ-editor/pull/356

LightTemplar, 2022-11-20:
I have experimented a little and found exact behavior:
If I create a new file, write there some non-Latin symbols and save it - it saves correctly. And I can successfully edit it also further. But, If I create a new file, write there only Latin symbols, save it and then try to add non-Latin symbols, I get exact this error. 
So, I think, https://github.com/CadQuery/CQ-editor/pull/356 doesn't work and https://github.com/CadQuery/CQ-editor/issues/276 isn't solved.

jmwright, 2025-01-29:
Potentially related to #463

jmwright, 2025-02-07:
See #480

closed 2025-02-08 by jmwright