CadQuery/CQ-editor #382: Can't install with Anaconda on macOS

revolter, 2022-11-19:
Running

```sh
conda install -c cadquery -c conda-forge cq-editor=master
```

fails with

```
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.

PackagesNotFoundError: The following packages are not available from current channels:

  - cq-editor=master

Current channels:

  - https://conda.anaconda.org/cadquery/osx-arm64
  - https://conda.anaconda.org/cadquery/noarch
  - https://conda.anaconda.org/conda-forge/osx-arm64
  - https://conda.anaconda.org/conda-forge/noarch
  - https://repo.anaconda.com/pkgs/main/osx-arm64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/r/osx-arm64
  - https://repo.anaconda.com/pkgs/r/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.
```

---

Cloning the repo and running

```sh
conda env create -f cqgui_env.yml -n cqgui
```

fails with

```
Collecting package metadata (repodata.json): done
Solving environment: failed

ResolvePackageNotFound:
  - cadquery=master
```

adam-urbanczyk, 2022-11-19:
Currently no arm64 builds are available.

revolter, 2022-11-19:
The `CQ-editor-MacOS.zip` from https://github.com/CadQuery/CQ-editor/releases/tag/0.2 works though, but I wanted the latest version.

andypiper, 2024-04-24:
Is there any chance of getting arm64 nightly builds for macOS? What would need to happen for that to become possible?