CadQuery/CQ-editor #385: Crash when using `.shell()` and `.fillet()`

BitLooter, 2022-12-21:
Running the following script causes a crash to desktop:
```
import cadquery as cq
result = (cq.Workplane("XY")
    .box(50, 50, 20)
    .faces("+Z")
    .shell(-1, kind="intersection")
    .edges()
    .fillet(1)
)
```

Terminal output follows:
```
Windows fatal exception: access violation

Thread 0x000019b8 (most recent call first):
  File "selectors.py", line 314 in _select
  File "selectors.py", line 323 in select
  File "asyncio\base_events.py", line 1823 in _run_once
  File "asyncio\base_events.py", line 570 in run_forever
  File "tornado\platform\asyncio.py", line 199 in start
  File "ipykernel\iostream.py", line 76 in _thread_main
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00001990 (most recent call first):
  File "threading.py", line 302 in wait
  File "threading.py", line 558 in wait
  File "IPython\core\history.py", line 829 in run
  File "IPython\core\history.py", line 58 in needs_sqlite
  File "<decorator-gen-17>", line 2 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Current thread 0x00000550 (most recent call first):
  File "cadquery\occ_impl\shapes.py", line 1931 in fillet
  File "cadquery\cq.py", line 1169 in fillet
  File "<string>", line 7 in <module>
  File "cq_editor\widgets\debugger.py", line 190 in _exec
  File "cq_editor\widgets\debugger.py", line 232 in render
  File "cq_editor\__main__.py", line 27 in main
  File "run.py", line 14 in <module>
```

I suspect this is due to the `fillet` value being physically impossible, though I doubt "hard crash to desktop" was the intended behavior. While this exact combination causes a crash, changing values of shell and fillet to other impossible values results in the incredibly unhelpful error message `StdFail_NotDone: BRep_API: command not done`. `kind` does not seem to make a difference.

CQ-editor version: 0.2
Windows version: Windows 10 Pro 22H2

BitLooter, 2022-12-21:
Just tested this on the latest development build, crash still occurs.

adam-urbanczyk, 2022-12-21:
I'm afraid that this is a limitation of the CAD kernel.

BitLooter, 2022-12-21:
Is there a more appropriate place to post this? Not sure from where in the stack this error would originate, I'm new to CadQuery and still not sure how everything is structured, I'd like to put this in the right place if possible.