CadQuery/CQ-editor #388: Sketch not visible or not there

mukral, 2023-01-22:
Hello,

I've tried the sketch-example
```
result = (
    cq.Sketch()
    .arc((0,0),1.,0.,360.)
    .arc((1,1.5),0.5,0.,360.)
    .segment((0.,2),(-1,3.))
    .hull()
   )
```
with several different cq-editor-packages (windows). Installation from artifacts like described.
But in no one (release version, development version from today (22.01.2023) or 1 month ago) a sketch is shown.
3d examplas work fine.

Do I have  to change something in preferences for 2D/sketch-visibility?

Thanks for Feedback
Markus

mukral, 2023-01-22:
Hi, workaround for me is the workplane integration of the sketch API:
```
import cadquery as cq
result = (
    cq.Workplane()
    .trapezoid(4,3,90)
    .sketch()
    .vertices()
    .circle(.5, mode='s')
    .reset()
    .vertices()
    .fillet(.25)
    .reset()
    .rarray(.6,1,5,1).slot(1.5,0.4, mode='s', angle=90)
    .finalize()
    .extrude(0.0001)
)
```

lorenzncode, 2023-01-22:
https://github.com/CadQuery/CQ-editor#showing-objects

_By default, CQ-editor will display a 3D representation of all Workplane objects_
...
_Note that show_object works for Shape and TopoDS_Shape objects too_
(and `show_object` also works with Assembly objects)

To visualize the Sketch in CQ-editor, you must select and extract a Shape(s).

Example to visualize edges:

```py
import cadquery as cq

result = cq.Sketch().segment((0, 0), (10, 0)).segment((5, 10))

show_object(result.edges().vals())
```

Example to show the face:

```py
import cadquery as cq

result = cq.Sketch().segment((0, 0), (10, 0)).segment((5, 10)).close().assemble()

show_object(result.faces().val())
```