CadQuery/CQ-editor #395: Release won't run on Ubuntu 22.04

atoav, 2023-03-31:
Steps to reproduce:
1. Download the [release 0.2](https://github.com/CadQuery/CQ-editor/releases/download/0.2/CQ-editor-Linux-x86_64.zip)
2. Unzip
3. `chmod +x CQ-editor.sh`
4. Run with `./CQ-editor.sh`

Step 4 fails and produces the following output:

```
[42288] mod is NULL - pyimod02_archiveTraceback (most recent call last):
  File "PyInstaller/loader/pyimod02_archive.py", line 32, in <module>
ModuleNotFoundError: No module named 'zlib'
[42288] mod is NULL - pyimod03_importersTraceback (most recent call last):
  File "PyInstaller/loader/pyimod03_importers.py", line 26, in <module>
ModuleNotFoundError: No module named 'pyimod02_archive'
Traceback (most recent call last):
  File "PyInstaller/loader/pyiboot01_bootstrap.py", line 17, in <module>
ModuleNotFoundError: No module named 'pyimod03_importers'
[42288] Failed to execute script pyiboot01_bootstrap
```

## System Details

```
lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.2 LTS
Release:        22.04
Codename:       jammy
```

```
python -V
Python 3.10.6
```

```python3 -V
Python 3.10.6
```

As a Python dev myself I tend not to install any python package outside of virtual environments. That means my systems python should be pretty much vanilla.

adam-urbanczyk, 2023-04-04:
I stopped with supporting pyinstaller TBH. @jmwright has a pipeline somewhere that produces pyinstaller bundles with the latest version. Maybe try that but YMMV.  #386 is the long term solution, but not ETA at this point of time.

adam-urbanczyk, 2023-06-03:
You could also try the nightly build: https://github.com/CadQuery/CQ-editor/releases/tag/nightly

closed 2025-02-11 by jmwright