CadQuery/CQ-editor #396: How to create a lof of objects quickly

atomsos, 2023-04-10:
I tried to create 250+ objects (half sphere and half cylinder), but It took almost 10 seconds, even with multithreading. It's too slow and unacceptable. 

I'm wandering if there is any way to create a lof of objects quickly, like 250 objs with at most 1 second.

I'm using Python 3.9 and OCP 7.7.0.0

atomsos, 2023-04-11:
besides, I found it's more time consuming to create a sphere than a cylinder.

adam-urbanczyk, 2023-04-11:
I'd start with sharing the code.

atomsos, 2023-04-15:
For simplification, it looks like this:
```python
import cadquery as cq
for i in range(150):
    sphere = cq.Workplane().sphere(1).translate([i, 0, 0])
    cylinder = cq.Workplane().cylinder(1, 1).translate([i, 0, 0])
    show_object(sphere)
    show_object(cylinder)
```

This code can be executed in CQ-editor directly, and it costs about 16 seconds on my laptop, (AMD Ryzen 5 4600H with Radeon Graphics).

atomsos, 2023-04-17:
Or maybe Cadquery is not a suitable tool for so many objects?
I'd like to show at most 1000 sphere and cylinders, for atoms structure show, do you have any recommendations?

adam-urbanczyk, 2023-04-17:
You can try something like this:
```python
import cadquery as cq

sphere = cq.Workplane().pushPoints([(i,0,0) for i in range(150)]).sphere(1,combine=False)
cylinder = cq.Workplane().pushPoints([(i,0,0) for i in range(150)]).cylinder(1, 1, combine=False)

show_object(sphere)
show_object(cylinder)
```

And maybe additionally adjust tessellation tolerance in the preferences. But on my (old) laptop the above code was already fast.

adam-urbanczyk, 2023-06-03:
OP seems to have lost interest. Best known answer provided, closing.

closed 2023-06-03 by adam-urbanczyk