CadQuery/CQ-editor #398: Strange debug behaviour on a freach install

lalebarde, 2023-04-16:
I have reinstalled cq-editor from scratch in a new conda environment:

```
mamba create -n cq python=3.11 pip
conda activate cq
mamba  install -c conda-forge -c cadquery cadquery=master # certainement inutile car dépendance de cq-editor
mamba  install -c conda-forge -c cadquery cq-editor=master
```
Note : mamba is fully compatible with conda, it is a rewrite in C far much faster.

The installed version is: 0.3.0dev
The debuging feature of cq-editor behave strangely: 

- breakpoints not respected
- commented code executed

It is like it keeps in memory some old versions of the code. If I close cq-editor and run it again, it works again as expected. After some use and code modifications, problems come back.

lalebarde, 2023-04-27:
Hi Adam, how can I test it please, I tried `mamba  install -c conda-forge -c cadquery cq-editor=debugger-ix` without success.

closed 2023-04-28 by adam-urbanczyk

adam-urbanczyk, 2023-04-29:
I merged the PR, you should be able to install via mamba (version is `master` and build string is `0.2_54_g2a9a02a`).