CadQuery/CQ-editor #400: cq-editor does not install from conda, apparently because the latest conda version is too new

wjodon, 2023-04-30:
On Windows 11, I have installed MiniForge from latest installer, and had no trouble activating the environment and installing cadquery, which passed its installation test.

However, the installation of cq-editor is not going well, as it is failing:
```
>conda install -c cadquery -c conda-forge cq-editor=master
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.

ResolvePackageNotFound:
  - conda==23.1.0
```
The conda version I am using seems to be newer:
```
>conda --version
conda 23.3.1
```
And while I have coded in Python and even made packages, Anaconda is only vaguely familiar. I searched for a hard dependency on conda 23.3.1 in this repo, but did not find such a thing.

Should I downgrade my version of conda as a workaround to enable installation?

adam-urbanczyk, 2023-04-30:
Is what you pasted complete? AFAICT in CI conda is not installed, so the issue is quite weird. BTW: nowadays I mostly use mamba.

adam-urbanczyk, 2023-06-03:
You can also try using our installer https://github.com/CadQuery/CQ-editor/releases/tag/nightly

zwn, 2024-02-16:
I was able to use micromamba to install cq-editor on windows 11 following the instructions in the readme.

jmwright, 2025-02-06:
@wjodon Please try again with the latest release of CQ-editor and reopen if the issue persists.

closed 2025-02-06 by jmwright