CadQuery/CQ-editor #410: Launch Viewport from python script

rohit-kumar-j, 2023-08-27:
I'm not sure if there is a way to already do this, but ...

Is there a way to launch only a viewport only showing parts of the GUI from within a python script?... without the toolbars, and other viewports and controls?

Something Like this:
```python
import cadquery as cq

length = 80.0 
height = 60.0  
thickness = 10.0
 
result = cq.Workplane("XY").box(length, height, thickness)

# Adding an option like this, opens a simple opengl viewer with the regular mouse movements
show_object(result, independent_viewport=true)
```
[image]

Instead  of this:
[image]

rohit-kumar-j, 2023-08-27:
Reason: I use [neovim](https://github.com/neovim/neovim) for programming, and do not want to switch to the full fledged editor. Setting up break points could be done outside the editor from within [neovim's debugger](https://github.com/rcarriga/nvim-dap-ui).

I would be glad to contribute a plugin for this,😄,... if anyone would guide me through cadquery's api, as I do not want to dig deep into cad query itself, and this is the first time working with PyQt

jmwright, 2023-08-27:
This is in the newer commits of CadQuery.

https://github.com/CadQuery/cadquery/pull/1366

rohit-kumar-j, 2023-08-27:
It opens a window but with errors:

```bash
PS C:\Koala\python\cadquery> python .\quick_window.py
2023-08-28 00:52:13.855 (  10.266s) [                ]vtkDemandDrivenPipeline:663    ERR| vtkCompositeDataPipeline (000001EF59A41FA0): Input port 0 of algorithm vtkTriangleFilter (000001EF5A129C40) has 0 connections but is not optional.
Traceback (most recent call last):
  File "C:\Koala\python\cadquery\quick_window.py", line 9, in <module>
    show(result)
  File "C:\Users\jsk71\AppData\Local\Programs\Python\Python311\Lib\site-packages\cadquery\vis.py", line 82, in show
    win.SetSize(*win.GetScreenSize())
TypeError: vtkRenderWindow.SetSize() argument after * must be an iterable, not NoneType
PS C:\Koala\python\cadquery>
```

Code:
```python
import cadquery as cq
from cadquery.vis import show

d = 10  # box is visible as expected
# d = 100  # empty window displayed; after 'r' key or zoom with middle mouse the box is displayed

result = cq.Workplane().box(d, d, d)

show(result)

```

rohit-kumar-j, 2023-08-27:
Let me mention it in https://github.com/CadQuery/cadquery/pull/1366

closed 2023-08-27 by rohit-kumar-j