CadQuery/CQ-editor #411: Do not run under latest micromamba/conda-forge

yomgui1, 2023-09-13:
On a Fedora 38, I've just installed latest micromamba and executed following commands:

`micromamba install -n base -c cadquery -c conda-forge cq-editor`

then

`micromamba run -n base cq-editor`

I've got following error:

```Traceback (most recent call last):
  File "/storage/projects/micromamba/bin/cq-editor", line 6, in <module>
    from cq_editor.__main__ import main
  File "/storage/projects/micromamba/lib/python3.11/site-packages/cq_editor/__main__.py", line 12, in <module>
    from .main_window import MainWindow
  File "/storage/projects/micromamba/lib/python3.11/site-packages/cq_editor/main_window.py", line 5, in <module>
    import cadquery as cq
  File "/storage/projects/micromamba/lib/python3.11/site-packages/cadquery/__init__.py", line 10, in <module>
    from .occ_impl.geom import Plane, BoundBox, Vector, Matrix, Location
  File "/storage/projects/micromamba/lib/python3.11/site-packages/cadquery/occ_impl/geom.py", line 5, in <module>
    from OCP.gp import (
ImportError: libTKIVtk.so.7.7: cannot open shared object file: No such file or directory
```

I've not found this lib (looks like to be in occt, but not found)

lorenzncode, 2023-09-14:
I reproduced the error.  It fails with ocp version 7.7.1.1dev.
```
 ocp                            7.7.1.1dev    py311_0              cadquery   
```

Try to specify the ocp version 7.7.1.0:
```
$ micromamba create -r ~/micromamba -n cqe
$ micromamba install -r ~/micromamba -n cqe -c cadquery cq-editor=master ocp=7.7.1.0

$ micromamba list -r ~/micromamba -n cqe 'cadquery|ocp|occt'
List of packages in environment: "/home/lorenzn/micromamba/envs/cqe"

  Name      Version  Build              Channel    
─────────────────────────────────────────────────────
  cadquery  master   2.3.1_45_gd780127  cadquery   
  occt      7.7.1    all_h16763a6_201   conda-forge
  ocp       7.7.1.0  py311h54fcf19_1    conda-forge
```

The GUI opens without error.
```
$ micromamba run -r ~/micromamba -n cqe cq-editor
```

lorenzncode, 2023-11-11:
This problem is resolved.  The dev OCP version is no longer installed unless the dev label is specified.
See also
https://github.com/CadQuery/cadquery/discussions/1422#discussioncomment-7350012

closed 2023-11-11 by lorenzncode