CadQuery/CQ-editor #412: 3DConnexion 3D Mouse Support

Spectre5, 2023-11-04:
Is there any change of getting support for 3D mice in CQ-Editor?  I have a 3D mouse from 3DConnexion that works great in FreeCAD via the spacenavd library.  It would be great to get it supported in CQ-Editor too.

jmwright, 2023-11-04:
@Spectre5 Is this the correct [link](https://spacenav.sourceforge.net/man_libspnav/) for the API?

PRs are always welcome. I don't know that the core team will have bandwidth or the interest to implement this anytime soon. I have a 3D mouse and have used it off and on, but there are just too many things that require focus in CadQuery-land. You can leave this issue open and other people can register their interest, and hopefully someone will pick it up eventually.

Spectre5, 2023-11-04:
Totally understandable.  I'h probably be somewhat out of my wheelhouse trying to implement that functionality but if I find some time I will look into it.  And yes, spacenavd used libspnav, but I think the editor could use a higher level interface - the output of spacenavd itself is usually used, I believe.

justbuchanan, 2023-11-05:
I may take a look at implementing this since I have SpaceNavigator mouse as well and would be interested in using it with cq-editor.

A quick search shows two python packages for interfacing with spacenav:

* https://github.com/johnhw/pyspacenavigator
  * Windows only :(
  * updated last year :+1: 
* https://spnav.readthedocs.io/en/latest/
  * it's still on pypi, but it hasn't been updated since 2011 and the repo on bitbucket doesn't exist anymore :(

justbuchanan, 2023-11-05:
This one looks promising: https://github.com/JakubAndrysek/pyspacemouse. Works on Windows, Linux, and Mac.

justbuchanan, 2023-11-05:
I sent https://github.com/CadQuery/CQ-editor/pull/413 as an initial proof-of-concept using `pyspacemouse`. It needs a fair amount of work, but I got some of the basics working.