CadQuery/CQ-editor #415: windows CQ-editor release does not support assembly.solve()

ShadowGuan, 2023-11-27:
[image]
[image]
[image]

The build was downloaded from the github CQ-editor latest build.
https://github.com/CadQuery/CQ-editor/releases/tag/nightly

I am running the sample assembly code in cadquery official webpage.
[https://cadquery.readthedocs.io/en/latest/assy.html](url)

And, for any type of assembly, the calling on solve() will lead to the same issue.

ShadowGuan, 2023-11-27:
it looks like some python modules are missing.
But have you guys ever tested the window exe?

I tried it on my two different PCs and both got the same issue.

lorenzncode, 2023-11-28:
I can reproduce the issue.

To workaround the problem, edit the `run.bat`.  Add the second line to append to PATH:
```
@echo off
set PATH=%~dp0\Library\bin;%PATH%
start /B Scripts/CQ-editor.exe
```

ShadowGuan, 2023-11-28:
> I can reproduce the issue.
> 
> To workaround the problem, edit the `run.bat`. Add the second line to append to PATH:
> 
> ```
> @echo off
> set PATH=%~dp0\Library\bin;%PATH%
> start /B Scripts/CQ-editor.exe
> ```

so great! The workaround works.

Thank you, @lorenzncode

closed 2025-03-09 by jmwright